1technophile/OpenMQTTGateway #454: OMG Lora receiving only 20% of packets sent to it

svwhisper, 2019-09-11:
I've now had many days using the Lora gateway and am finding that only 20% of packets sent by a remote node are received by the gateway.  This isn't a range issue, as I moved the sender within a few feet of the gateway and didn't see any improvement. The antenna on the gateway is a commercial-grade 433mHz external antenna.  I coded the receive example in Mistry's repo and ran it on a Moteino.  It received 100% of the packets.

I've verified that all the Lora parameters (freq, bandwidth, spreading factor, etc.) are aligned between nodes.  I wonder if the gateway is busy doing something else when the packets arrive?  I only have Lora functionality enabled, i.e. no other modules like BLE.

I will try to progress root cause through review of the code but wanted to report this now.

1technophile, 2019-09-11:
thanks for this report.
could you indicate what is the sender board and firmware please?

Cloolalang, 2019-10-08:
Hey Florian, 
Congratulations on this great project!
I have a Lora question:
I have started with and ESP-32 and a RA-02 433MHz lora module. I have installed V0.9.2 Binary ok
Reading the guide:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-LORA
I see that we can setup the transmitter parameters for the Lora transmissions.
Is is possible to configure my Lora receive parameters as well?
The reason I ask is that the default seems to be 868MHz, and mine needs to be set to 433MHz.
I would really like to be able to do this via an MQTT command rather than re-compiling the ESP firmware.
Best Wishes
Patrick

Cloolalang, 2019-10-08:
Hey, So After more research, I have it all working except I can only get the ESP32+RA-02 working both transmit and receive using the Lora.h library version 3.0.\

closed 2019-12-16 by stale[bot]

nikito7, 2023-06-18:
Need to fix node to single channel