1technophile/OpenMQTTGateway #455: Configuration with Wifi Manager

opl-git, 2019-09-13:
Hi,
thank you very much for keep working on this awesome piece of software.
I downloaded the newest version of Openmqtt and flashed it, using the espressif ESP32 Downloader software.
After that I logged in to the AP (OPENMQTT_Gateway) and configured my wifi and mqtt broker.
The ESP32 connects to my wifi without problems.
After that I can't connect to the ESP32 directly using it's IP adress... is this normal??
I thought I could configure it more (for example the mqtt topic) once it is connected to my wifi.

Kind regards
Daniel

1technophile, 2019-09-13:
Hi,

At this time there is no extra configuration available once connected to the wifi/broker (except ble [interval](https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-bluetooth-BLE-09#setting-the-time-between-scans-and-force-a-scan)).
Could you indicate which configuration you would like to do?

opl-git, 2019-09-14:
Ok sorry, my fault. 😄 
I think the possibility to change some config stuff would be nice, but not essential like
- change the topic of the mqtt data
- change the mqtt data (like the server adress, user, pass), right know it is only possible to flash it again and be able to change that, or?

1technophile, 2019-09-14:
Yes you can flash it again, just do an erase all before and you will start with a fresh new esp.

botts7, 2019-10-26:
As I am trying to set up multiple devices and have no luck compiling my own bin files. having the ability to change MQTT topic once up and running would be a great enhancement
Loving the project!

botts7, 2019-11-04:
any chance on releasing a pre compiled beta of current state of development release?

1technophile, 2019-11-04:
you mean releasing binaries from the development branch without this issue?

botts7, 2019-11-04:
if possible release binary with any working enhancements from previous  official release. as I am unable to get my own bin compiled for my boards. A pre milestone release? 🙄 looking forward to the next release either way! cheers!

1technophile, 2019-11-08:
> if possible release binary with any working enhancements from previous official release. as I am unable to get my own bin compiled for my boards. A pre milestone release? 🙄 looking forward to the next release either way! cheers! (ble version)

See below:
https://github.com/1technophile/OpenMQTTGateway/releases/tag/v0.9.3beta

botts7, 2019-11-09:
Legend! thank you!

1technophile, 2019-12-06:
done

closed 2019-12-06 by 1technophile