1technophile/OpenMQTTGateway #456: MQTT - Socket error

marrold, 2019-09-19:
**Describe the bug**

My OMG Node is occasionally losing connection to the Mosquitto Broker temporarily. In the logs I see `Socket error on client omg-kitchen, disconnecting`

A Wireshark trace shows a "Malformed Packet" sent from the node, followed by a RST from the Broker.

**To Reproduce**

Connect ESP32 to Broker and wait for a disconnect.

**Expected behavior**

I wouldn't expect such frequent disconnects

**Screenshots**
[image]

**Environment (please complete the following information):**

Hass.io Version: 188
Home Assistant Version:  0.98.5.
Mosquitto Add-on Version: 5.0
OMG Release: 0.9.2
Board: ESP32
Modules: BLE only (0.9.2)

**Additional context**

[Packet Capture](http://cdn.marrold.co.uk/files/Stuff/omg-mqtt.pcapng)

marrold, 2019-09-20:
I've just flashed OMG to another ESP32 with an almost identical config (Except Gateway name) and it seems more stable, so this could be a Hardware issue of some sort. 

I've ordered another one to do some more testing.

Thanks

1technophile, 2019-09-20:
Ok, thanks for the feedback I will wait for your results before digging into this.

marrold, 2019-09-25:
I have disabled `home_presence` and left it over night but unfortunately the issue still occurs. In this example the MQTT Topic Length is a very high number: 

[Invalid Topic Length](cdn.marrold.co.uk/files/Stuff/omg-mqtt.pcapng)

closed 2020-01-31 by stale[bot]