1technophile/OpenMQTTGateway #457: Fitbit Alta HR crashes OMG

marrold, 2019-09-20:
**Describe the bug**
Fitbit Alta HR crashes OMG

**To Reproduce**
1. Place Fitbit Altra HR near ESP32 BLE Node
2. Wait for OMG to crash (almost immediately) 

**Logs**

```
Creating BLE buffer
device detected
D46703B52CD4
BLErssi
-49
txPower
5
BLE DISTANCE :
ovf
Pub json into:
omg/home_presence/omg-kitchen
{"id":"d4:67:03:b5:2c:d4","name":"Alta HR","rssi":-49,"txpower":5,"distance":8.170728e9}
Pub data per topic
id
d4:67:03:b5:2c:d4
name
Alta HR
rssi
-49
txpower
5
distance
ovf

Stack smashing protect failure!

abort() was called at PC 0x401589cf on core 0

Backtrace: 0x40092244:0x3ffe5bd0 0x40092475:0x3ffe5bf0 0x401589cf:0x3ffe5c10 0x400d2035:0x3ffe5c30 0x400d4261:0x3ffe5c60 0x400d4c39:0x3ffe5fb0 0x400d60a3:0x3ffe5ff0 0x400da4dd:0x3ffe6370 0x400d971a:0x3ffe6450 0x400e9731:0x3ffe64a0 0x400e722e:0x3ffe64c0 0x4008e189:0x3ffe64f0

Rebooting..

```

**Environment (please complete the following information):**
Hass.io Version: 188
Home Assistant Version: 0.98.5.
Mosquitto Add-on Version: 5.0
OMG Release: 0.9.2
Board: ESP32
Modules: BLE only (0.9.2)

**Additional context**

- It does not crash when scanning is passive: `pBLEScan->setActiveScan(false);`
- It does crash when `pubBLEServiceData` is `false`

Thanks!

1technophile, 2019-09-20:
Thanks for this detailled report,
Unfortunately I didn't found a lot of documentation about TXPower which seems to be the issue here,
as a bypass I propose you to replace the '==' by '>=' [here](getTXPower)

```C++
  if (txPower >= 0)   txPower = -59; 
```

marrold, 2019-09-20:
Thanks for the quick suggestion, I have made the proposed edit and can confirm the issue is resolved.

1technophile, 2019-09-20:
Update of the code made

closed 2019-09-20 by 1technophile