1technophile/OpenMQTTGateway #460: compilation error IR

clubbit, 2019-09-21:
Good afternoon, it says that the error is in line 111 "ZgatewayIR"

String hex = resultToHexidecimal(&results);

I use libraries from your set.

Arduino: 1.8.1 (Windows 10), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino: In function 'boolean reconnect()':

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:151:57: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       pub(will_Topic,Gateway_AnnouncementMsg,will_Retain);

                                                         ^

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:151:57: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:153:48: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       pub(version_Topic,OMG_VERSION,will_Retain);

                                                ^

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:153:48: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino: In function 'void stateMeasures()':

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:763:45: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

       pub(subjectSYStoMQTT,JSONmessageBuffer);

                                             ^

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino: In function 'void pub(char*, long int)':

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:1135:31: warning: conversion lacks type at end of format [-Wformat=]

     sprintf(val, "%l", payload);

                               ^

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:1135:31: warning: too many arguments for format [-Wformat-extra-args]

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:1135:31: warning: conversion lacks type at end of format [-Wformat=]

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:1135:31: warning: too many arguments for format [-Wformat-extra-args]

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino: In function 'void pub(char*, double)':

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:1141:31: warning: format '%d' expects argument of type 'int', but argument 3 has type 'double' [-Wformat=]

     sprintf(val, "%d", payload);

                               ^

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\OpenMQTTGateway_v092.ino:1141:31: warning: format '%d' expects argument of type 'int', but argument 3 has type 'double' [-Wformat=]

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\ZgatewayIR.ino: In function 'void IRtoMQTT()':

ZgatewayIR:111: error: 'resultToHexidecimal' was not declared in this scope

       String hex = resultToHexidecimal(&results);

                                                ^

C:\Users\xpiri\Documents\Arduino\OpenMQTTGateway_v092\ZgatewayIR.ino:148:35: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         pub(subjectIRtoMQTT,IRdata);

[... 423 lines cut ...]

         if (valueBITS == 0) valueBITS = kTcl112AcStateLength;

                                         ^

ZgatewayIR:796: error: 'class IRsend' has no member named 'sendTcl112Ac'

         irsend.sendTcl112Ac(dataarray, valueBITS, valueRPT);

                ^

ZgatewayIR:802: error: 'kTecoBits' was not declared in this scope

         if (valueBITS == 0) valueBITS = kTecoBits;

                                         ^

ZgatewayIR:803: error: 'class IRsend' has no member named 'sendTeco'

         irsend.sendTeco(data, valueBITS, valueRPT);

                ^

exit status 1
'resultToHexidecimal' was not declared in this scope

clubbit, 2019-09-21:
updated the library to the latest version. everything worked

1technophile, 2019-09-22:
Good !

closed 2019-09-22 by 1technophile