1technophile/OpenMQTTGateway #461: WiFi does not reconnect after disconnection

marrold, 2019-09-21:
**Describe the bug**
After a WiFi Disconnection, OMG Node does not attempt to reconnect, only printing `wifi disconnected` to serial.

**To Reproduce**

1. Allow OMG Node to connect to WiFi / MQTT and start publishing data.
2. Force WiFi disconnect by turning off Access Point or Blocking Client.
3.  Re-enable WiFi
4. Serial Output still shows `wifi disconnected` without attempting reconnection.

**Expected behavior**
OMG Node should attempt to reconnect to WiFi on disconnection. After several re-tries, it should reboot.

** Fix **

Add the following below `trc("wifi disconnected");` in `main.ino` - I would submit a PR but maybe you want to do it differently: 

```
    #if defined(ESPWifiManualSetup)
      setup_wifi();
    #else
      setup_wifimanager(false);
    #endif
```

**Environment (please complete the following information):**

Hass.io Version: 188
Home Assistant Version:  0.98.5.
Mosquitto Add-on Version: 5.0
OMG Release: 0.9.2
Board: ESP32
Modules: BLE only (0.9.2)

**Additional context**

Apologies for raising a few issues these last few days, I am setting up OMG for the first time and have stumbled across these issues whilst testing / looking at the code. Thanks for all your hard work.

1technophile, 2019-09-22:
Hello,
@h4nc raised a similar issue with esp32 and apple router. It seems to be a bad wifi status return from esp32 wifi library. Could you indicate the brand and the model of your router please ? 

https://community.openmqttgateway.com/t/esp32-doesnt-stay-online/570/59?u=1technophile

marrold, 2019-09-22:
Hi, the router is a [Mikrotik hAP ac](https://mikrotik.com/product/RB962UiGS-5HacT2HnT) - I've been having a similar issue to @h4nc, specifically `errno: 113, "Software caused connection abort"` However I'm still trying to get a packet capture to see what is happening on the network level - the router says the device is still connected. Once I've got more evidence I'll reply to that thread.

This issue was more specifically aimed at the lack of a WiFi reconnect function in OMG - is this handled transparently by the WiFi library in theory?

marrold, 2019-09-22:
Just a quick update whilst I explore this issue, it seems like `client.connect()` is hanging when Wifi is down, it never exits to increase the counter. Sounds like this [issue with pubsubclient](https://github.com/knolleary/pubsubclient/issues/624)

I've upgraded `esp32_platform` to `espressif32@1.10.0` as the change list lists lots of WiFi fixes, and now client.connect() exits immediately with an error when the Wifi is disconnected.

Not quite a fix, but progress.

marrold, 2019-09-23:
@1technophile ignoring this issue with specific routers, should the Wifi automatically reconnect after a disconnection? 

In the code I can see the following `trc("wifi disconnected");` but there's no attempt to reconnect. Are reconnections handled by the library, or should the OMG code attempt to reconnect? 

Thanks

1technophile, 2019-09-23:
Even if the wifi is disconnected OMG should retry after due to the loop mechanism.
It is possible maybe that the disconnected status of the wifi doesn't generate a disconnect status of the MQTT client.
As I'm first checking the status of the MQTT client maybe we have a track to follow.
I will try to revert the order of the checks between MQTT and Wifi.

marrold, 2019-09-23:
On line `850` in `main.ino` I see an if statement:
`if (WiFi.status() == WL_CONNECTED){`

The else is:
```
        }else{
          trc("wifi disconnected");
        }
```
So if the Wifi is disconnected it doesn't enter the `reconnect()` function or ever restart?

1technophile, 2019-09-24:
If wifi is disconnected it goes to the loop and go to 
```C++
  if (!client.connected()) { // not connected
```
This method checks if the MQTT client is connected but we should better recheck the wifi first, it is possible that this method, in case of no wifi with your configuration block the main loop (Mine reconnects properly).

I will propose a new algorythm to test

wimpie007, 2019-10-03:
any news about this? having problems with this issue also.

1technophile, 2019-10-04:
it is in progress

wimpie007, 2019-10-04:
i got this as error this morning after wifi went down.
https://pastebin.com/L9guQ68B

1technophile, 2019-10-07:
Could you try to replace [here](
https://github.com/1technophile/OpenMQTTGateway/blob/e4de45f2194fa1fe30b2f4e51d6acdc1925802be/platformio.ini#L91)
`esp32_platform = espressif32@1.9.0`
by
`esp32_platform = espressif32@1.10.0`

h4nc, 2019-10-08:
@1technophile I already tried this. Unfortunately the ESP32 still doesn't reconnect after connection loss (ESP8266 does). ESP32 still needs to be unplugged.

wimpie007, 2019-10-08:
@1technophile did that already. also with 1.11.0.
It seems a bit different: with 1.9.0 the ESP32 would "halt". with 1.10.0 and 1.11.0, it would reboot wich is better. (it seems the crash is not always related to wifi!!! so maybe not in right 'issue')

1technophile, 2019-10-11:
I can confirm that with 1.10.0 and 1.11.0 the wifi disconnection is better recognised.
Nevertheless the reboot that occurs after is difficult to explain.

I have done some modifications there, if you could test them it would be great:
https://github.com/1technophile/OpenMQTTGateway/tree/connection-mng

1technophile, 2019-10-11:
```

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
```
Does this occurs always at the same time in the code, after:
```
OpenMQTTGateway ip: 
192.168.1.50
```

h4nc, 2019-10-11:
I erased flash and refreshed the same firmware and now it works. Probably my fault. Will delete the post above, as it makes the thread longer that it has to be. 

Will keep ESP32 and ESP8266 powered on to see how the perform now and come back with the results.

h4nc, 2019-10-12:
The ESP8266 is connected since some hours know. 

The thing is it seems like it opens an access point simultaneously. Because I can see an AP with the last mac address digits of that device in my network.
Maybe a bug in the new firmware?

1technophile, 2019-10-12:
> The thing is it seems like it opens an access point simultaneously. Because I can see an AP with the last mac address digits of that device in my network.

I'm able to reproduce the issue on ESP8266

1technophile, 2019-10-12:
Corrected [here](https://github.com/1technophile/OpenMQTTGateway/tree/connection-mng)

h4nc, 2019-10-12:
Was this only a ESP8266 issue or should I also update the ESP32

1technophile, 2019-10-13:
The modification of code apply to both board. I have only seen the AP in the ESP8266.
But as the modification impact both I would test it on both types.

h4nc, 2019-10-14:
I'd like to confirm that I don't see the AP any more after the update. Thanks

closed 2019-11-02 by 1technophile