1technophile/OpenMQTTGateway #462: Pilight build not working?

klockie86, 2019-09-22:
Hello,

First of all, what as nice project, very easy to setup. Flashing my nodemcu was done in a few seconds.

However I'm having a small issue with a pilight build.
I've forked the development branch, then did my settings in User_config.h. 
I removed the comment on "#define ZgatewayPilight "Pilight" //ESP8266, Arduino, ESP32".
Then I compiled the project with "PIO Upload (nodemcuv2-rf)"

After setting up wifi and mqtt, after booting the following message appears in the serial console:
`Connecting to MQTT by IP adress
xxx.xxx.xxx.xxx
RF_EMITTER_PIN
4
RF_RECEIVER_PIN
0
RF setup ok
**ZgatewayPilight setup cannot be done, comment first ZgatewayRF && ZgatewayRF2 &&** ZgatewayRF315
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
client disconnected
MQTT connection...
Connected to broker
`
However I have the ZgatewayRF & ZgatewayRF2 commented, ZgatewayRF315 is not pressent at all in my user_config.

klockie86, 2019-09-22:
Nevermind, i used the wrong platformio.ini file. So I didn't have the "PIO Upload (nodemcuv2-pilight option)"

closed 2019-09-22 by klockie86

TopdRob, 2019-10-09:
I'm having this issue as well. I was wondering what you exactly did to solve it?

klockie86, 2019-10-10:
For building use; "nodemcuv2-pilight". This includes the right libraries.