1technophile/OpenMQTTGateway #464: Bad values for moisture - Mi Flora

leonols, 2019-09-29:
**Describe the bug**
Values fo moisture are bad, for other characteristics they are good.
I noticed that rest_data for moisture is 2 characters shorter

```
Creating BLE buffer
device detected
C47C8D65D539
BLErssi
-75
txPower
-59
BLE DISTANCE :
5.83
Pub json into:
home/home_presence/OpenMQTTGateway_ESP32_BLE
{"id":"c4:7c:8d:65:d5:39","name":"Flower care","rssi":-75,"distance":5.832737}
Get services data :
1
Pub json into:
home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/C47C8D65D539
{"id":"c4:7c:8d:65:d5:39","name":"Flower care","rssi":-75,"distance":5.832737,"servicedata":"712098002b39d5658d7cc40d0910028b01","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
Processing BLE device data
mi flora data reading
Creating BLE buffer
rest_data
712098002b39d5658d7cc40d0910028b01
data_length
4
395.00
Pub json into:
home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/C47C8D65D539
{"fer":395}
Processing BLE device data
Creating BLE buffer
device detected
C47C8D65CA17
BLErssi
-73
txPower
-59
BLE DISTANCE :
4.76
Pub json into:
home/home_presence/OpenMQTTGateway_ESP32_BLE
{"id":"c4:7c:8d:65:ca:17","name":"Flower care","rssi":-73,"distance":4.756481}
Get services data :
1
Pub json into:
home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/C47C8D65CA17
{"id":"c4:7c:8d:65:ca:17","name":"Flower care","rssi":-73,"distance":4.756481,"servicedata":"712098009917ca658d7cc40d08100111","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
Processing BLE device data
mi flora data reading
Creating BLE buffer
rest_data
712098009917ca658d7cc40d08100111
data_length
2
1.00
Pub json into:
home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/C47C8D65CA17
**{"moi":1}**
```

The correct moisture value for the above example is 17

i'm use:
OpenMQTTGateway version 0.9.2
Mi Flora version 3.1.9

kanttti, 2019-09-29:
I have also noticed the same and opened a pull request with a fix to this issue.

For me it seems this issue is NOT present in v0.9.2, but only in the currrent head of development. 
Which commit did you use to compile the code or did you use the compiled binaries?

leonols, 2019-09-29:
I’m use Latest commit d2dd613
With Your fix works fine.
Thanks

closed 2019-09-29 by leonols