1technophile/OpenMQTTGateway #467: RF transmitter protocol 17

mihc, 2019-10-04:
Hi,

is it possible to add support to more RF protocols? I would like to replicate signal from home thermostat which controls IR heater (ON/OFF function). 
Thermostat sends data on protocol 17. This is what i get on OMG RF 433MHz receiver part:
b'{"value":142465,"protocol":17,"length":24,"delay":737}'
b'{"value":142465,"protocol":17,"length":24,"delay":734}'
b'{"value":142465,"protocol":17,"length":24,"delay":732}'
OR
b'{"value":142466,"protocol":17,"length":24,"delay":732}'
b'{"value":142466,"protocol":17,"length":24,"delay":734}'
b'{"value":142466,"protocol":17,"length":24,"delay":731}'
etc..

If i try to send data to MQTTto433; payload: {"value":142465,"protocol":17,"length":24,"delay":731}
it sends by protocol 1 and IR thermostat does not accept it.

Thanks and best regards

1technophile, 2019-10-04:
Hi,

Could you post the serial monitor output please?

mihc, 2019-10-04:
Hi,

thank you for fast reply. Here is serial output for different states:

Idle - state off. I guess it sends RF data once a minute:
[ir-panel-state-off.txt](https://github.com/1technophile/OpenMQTTGateway/files/3692128/ir-panel-state-off.txt)
When it is in on state, RF data once a minute:
[ir-panel-state-on.txt](https://github.com/1technophile/OpenMQTTGateway/files/3692125/ir-panel-state-on.txt)
When i powered off:
[ir-panel-power-off.txt](https://github.com/1technophile/OpenMQTTGateway/files/3692126/ir-panel-power-off.txt)
When i powered on:
[ir-panel-power-on.txt](https://github.com/1technophile/OpenMQTTGateway/files/3692127/ir-panel-power-on.txt)