1technophile/OpenMQTTGateway #468: ESP32 with AI Thinker Lora module RA-02 only working with Lora.h version 3.0.0

Cloolalang, 2019-10-12:
I decided to build my own Lora boards for the OMG Lora.
Im using ESP32 Dev board with the RA-02.
There seems to be an issue with this module only working with the earlier verions of the Lora.h library.
I admit, that the RA-02 is not listed as a supported device, however it all works fine with version 3.0.0 of the lib.
Reported here:
https://github.com/sandeepmistry/arduino-LoRa/issues/296

I had difficulty finding out what version of the Lora libs are complied inside the releases, Please would you list the version of the dependent libs on the wiki?

If there is a change that you can make a .bin available with the version 3.0.0 Lora.h that would really help as that you can add another device to the list.

Cheers
P

1technophile, 2019-10-12:
Hello,

you can find the libraries list [here](https://github.com/1technophile/OpenMQTTGateway/blob/4342a1521c5c48bc99eba0ea8e4144e61ca154c6/platformio.ini#L54)

When I check the library [list](https://github.com/sandeepmistry/arduino-LoRa/releases), I can't find a 3.0 version

Cloolalang, 2019-10-12:
Hi Florian, 
Thanks for your response.
My apologies, Typo, I mean the Lora.h 0.3.0 Library.
I will setup on PlatformIO next, as its seems the best way to tailor the build for my needs, 
Many Thanks

closed 2020-02-15 by stale[bot]