1technophile/OpenMQTTGateway #469: Is lib meant to be empty?

crzg, 2019-10-13:
The lib folder for 0.9.2 is empty aside from a readme (https://github.com/1technophile/OpenMQTTGateway/tree/development/lib)
0.9.1 has 19 libraries

Is this on purpose?

1technophile, 2019-10-13:
Hello,

Yes, with v0.9.2 we now use platformio which download automaticaly the appropriate library.
They are listed below : 
https://github.com/1technophile/OpenMQTTGateway/blob/4342a1521c5c48bc99eba0ea8e4144e61ca154c6/platformio.ini#L54

1technophile, 2019-10-18:
I close the issue, if you have further question don't hesitate to go to the community:
https://community.openmqttgateway.com/latest

closed 2019-10-18 by 1technophile

Koen00, 2019-10-24:
If I may hook into this thread, I am trying to flash my Arduino mega with the PlatformIO but having always failed checkout attempts when installing. So PlatformIO is not going to work for me. Also unable to change the board from esp8266 to the megaatmega2560. 
You say that the arduinoIDE can be used, but this is not useful since the appropriate libraries are not available.
Starting to get very frustrated, trying to get it to work for two weeks now.

1technophile, 2019-10-24:
Hello,

You should better ask support/questions on the forum rather than here:
https://community.openmqttgateway.com/

gclarkedublin, 2019-10-24:
In fairness, should Arduino IDE be removed as an option altogether and fron any documentation to avoid confusion? I just used an older version but this change was not clear, and a few parts of the Wiki continue to show Arduino IDE screenshots

1technophile, 2019-10-24:
Idealy we should write a detailled tutorial for uploading with arduino ide. Regarding the wiki, once the new docs http://docs.openmqttgateway.com are finished I will delete it.