1technophile/OpenMQTTGateway #475: BLE Maximum Distance/RSSI

maxi1134, 2019-10-28:
**Is your feature request related to a problem? Please describe.**
My location often switch from Bedroom(In which I am located) to Bathroom. Since the ESP32 in the bathroom can detect my watch from there

**Describe the solution you'd like**
A way to use the RSSI as a filter. Per instance only show Beacons within a certain reach.
This would make it harder to get false results when using more than a couple ESP32s as beacons receivers.

**Describe alternatives you've considered**
I tried using a different project but would rather stick with OpenMQTTGateway

**Additional context**

1technophile, 2019-10-28:
Seems easy to implement.
Could you point me an idea of RSSI you get from the bedroom and from the bathroom from the same device.
If we set a min value at the build time per default do you think it will answer your needs?

maxi1134, 2019-10-28:
I just took a quick look and It seems to have between -80 and -95 when I am in the same room as the ESP32.

So anything under -95 could be ignored in my case.

Something at the build time would work well. But perhaps something like an MQTT topic to set the limit would allow for better tuning without having to reflash the ESP everytime you need to adjust.

BTW, my beacon is an unpaired Fitbit.

1technophile, 2019-11-03:
see https://github.com/1technophile/OpenMQTTGateway/pull/482

1technophile, 2019-11-08:
Discussion on #482

closed 2019-11-08 by 1technophile