1technophile/OpenMQTTGateway #477: Mertik Maxitrol RF protocol

PaulAntonDeen, 2019-11-01:
**Is your feature request related to a problem? Please describe.**
I'm trying to automate my fireplace by using a ESP2866 with RF modules to replicate the signal from the remote. I'm trying to listen to the signals from the remote but I get the same values for every button I push. My guess is that since the remote needs to be paired to the receiver that the first part is its identifier and the payload is the next part of the signal. However OMG only echo's the identifier that looks like this:
{
    "value": 4194303,
    "protocol": 29,
    "length": 22,
    "delay": 679
}

**Describe the solution you'd like**
Get the whole signal and be able to output back to the receiver

PaulAntonDeen, 2019-11-04:
I was able to brute force hack and guess my way to enough of the commands.

closed 2019-11-04 by PaulAntonDeen

akasma74, 2019-11-04:
Could you share your solution please?

PaulAntonDeen, 2019-11-04:
My fireplace is a Faber Gas fireplace with the Mertik remote

These are the commands I'm using.
>Minimize and kill flame: {"value":2043326,"protocol":17,"length":22,"delay":367}
Decrease flame (about 10 steps between max and min)  {"value":2043326,"protocol":5,"length":22,"delay":367}
Minimize but leave on: {"value":2043316,"protocol":5,"length":22,"delay":367} Maximize {"value":2043316,"protocol":26,"length":22,"delay":367}

All of it implemented in Home Assistant with a NodeMCU as the 433Mhz transmitter.

If anybody ever finds the command to trigger ignition and do step increase the flame, I would be grateful!

akasma74, 2019-11-04:
So basically you played with different "value" and "protocol" and came up with these commands?
Wonder where you started from as they're all different to what your remote sends.

PaulAntonDeen, 2019-11-04:
I started with a second NodeMCU as the listener device and just pressing buttons on my remote and trying them on the send device. I got two of these from that.  As I could not capture the rest of the codes that way I just started varying the protocol from the captured values. Delay has an impact as well but it seems that all the ones that worked had a delay around this value so I settled on that.

1technophile, 2019-11-04:
Thanks for the method sharing!