1technophile/OpenMQTTGateway #485: GPIOKeyCode what is it?

mpeirone, 2019-11-04:
Hi! I can't find any documantations about GPIOKeyCode,
what is it?

1technophile, 2019-11-05:
Hi, here is the definition:
    This reads up to 4 gpio with latch signal (gpio line) and combines the bits as KeyKode - can be connected
    to EV1527 receiver.

If you have more questions maybe @McGr3g0r can answer

1technophile, 2019-11-08:
I close the issue , if you have other question don't hesitate to go to community.openmqttgateway.com

closed 2019-11-08 by 1technophile

McGr3g0r, 2019-12-10:
> Hi, here is the definition:
> This reads up to 4 gpio with latch signal (gpio line) and combines the bits as KeyKode - can be connected
> to EV1527 receiver.
> 
> If you have more questions maybe @McGr3g0r can answer

Sorry for late reply, I guess we all need time dilation machine. 
I can confirm that GPIOKeyCode reads  specific gpio values to be represented  as byte value which is published to mqtt. 

In my use case the code is read from EV1527 (usually 4-bits) when the latch bit is active : see definitions GPIOKeyCode_D0_PIN to GPIOKeyCode_D3_PIN and GPIOKeyCode_LATCH_PIN.
and EV1527 receiver in the image - the VT signal is a latching signal:

[ EV1527 Receiver ](https://www.dhresource.com/0x0/f2/albu/g5/M00/85/25/rBVaI1lAme6ALkAmAAKyYu2c728215.jpg)

[ My gateway](https://photos.app.goo.gl/hPwYj9YNohJQsTgK9)

Best Regards,
McGr3g0r