1technophile/OpenMQTTGateway #486: Hassio Add-on : Mosquitto Broker (MQTT) - Auto Discovery not working

rawframe, 2019-11-04:
Hello,
Good work! 
Can you help me understand why Home Assistant (Hassio with Mqtt add-on) is **not auto discovering**? I use other mqqt devices that auto discover.

The **warning messages** are like: 
**- "Unable to parse JSON XXmacaddressXXgatewayRF: 'home/OpenMQTTGateway_ESP8266_RF/433toMQTT'"
- "Unable to parse JSON XXmacaddressXXgatewayRF: '{{ value_json.value }}'**

In entity states i can see "binary_sensor.openmqttgateway_esp8266_rf" with state as on and "sensor.gatewayrf with no state. 
If I manually set up an an rf mqtt switch in configuration.yaml it does work in sending on/off rf.
Thanks in advance.

1technophile, 2019-11-05:
Hi,

Could you indicate the version of OMG and Hassio used?

rawframe, 2019-11-05:
Thanks for the reply.
Hassio 192 and OMG  092 with only RF module installed via Platformio.

closed 2020-01-11 by stale[bot]