1technophile/OpenMQTTGateway #487: Nodemcu v3 and Pilight problem

Saniksanik, 2019-11-06:
I have NodeMCU v3, cheap china receiver 433 mhz and Digoo Temp sensor R8H.
Two problems:
NodeMCU don,t save wifi settings with this firmware. I connect to nodemcu AP, settings saved but after reset device don,t connect to thr my home router. Only one after confirguration. 
On Tasmota firmware wifi work perfect on this board.
Can i improve that? I very need Pilight support

And second. I don,t see any packet from Digoo sensor. I write PIN 0 and connect receiver to D3 pin on nodemcu. Receiver has external 5V power and normally work with PC and serial adapter. Maybe on Nodemcu i must use other pin for receiver?

akasma74, 2019-11-06:
I think Digoo R8H is not supported by OMG yet.
R8S partially is, but it's a different device.

Saniksanik, 2019-11-06:
Sorry, my sensor is r8s

1technophile, 2019-11-24:
>And second. I don,t see any packet from Digoo sensor. I write PIN 0 and connect receiver to D3 pin on nodemcu. Receiver has external 5V power and normally work with PC and serial adapter. Maybe on Nodemcu i must use other pin for receiver?

You can try the pins listed below:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1864843263

closed 2020-01-31 by stale[bot]