1technophile/OpenMQTTGateway #491: Adding support for SmartDry sensor

DieKatzchen, 2019-11-12:
**Is your feature request related to a problem? Please describe.**
I have SmartDry dryer sensor I would like to use with OMG

**Describe the solution you'd like**
Add support for SmartDry to the BLE component

**Describe alternatives you've considered**
I started to roll my own and then found yours

**Additional context**
When it senses motion, the sensor wakes up and starts advertising. Upon connection, it will send its data on service "00003AAA-0000-1000-8000-00805F9B34FB" characteristic "0000AAA1-0000-1000-8000-00805F9B34FB".  It is 8 bytes, and seems to be two ints. The first 4 bytes appear to be the moisture reading and the second 4 appear to be the temperature. I think. I'm going by the fact that in the dryer the reading I think is moisture drops off suddenly, followed a minute later by the temperature dropping off, and then the sensor going to sleep. I believe the moisture reading is in AB5M since that's the only identifying mark on the IC.  The characteristic has notify, and the device will simply go to sleep after 2.5 minutes of no motion. That's how the system knows the dryer has stopped. i have some pictures of the board, but the whole thing is covered in potting compound except the moisture sensor and battery holder.

1technophile, 2019-11-12:
Hello,

This device is interesting, before thinking on making a connection to retrieve data, is there a way to get the data by service data advertising.
So as to check that you can do a scan with [Beacon simulator](https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=en) and check if there is Service data advertised. So as to see that the device must be disconnected from your phone.

Here is a post on whicvh I give [details](https://1technophile.blogspot.com/2017/11/mi-flora-integration-to-openmqttgateway.html) on the Mi Flora protocol reverse engineering.

DieKatzchen, 2019-11-13:
Well, I got some interesting data. I recorded some resting data, then I held it over the stove, then I waited until it was about to sleep and started scanning, waiting until it stops transmitting. I think I found the signal that it's about to sleep. 

`AE 57 A3 41 90 27 19 42 91 00 43 07 00
C0 2B A3 41 90 27 19 42 01 00 43 07 00
C0 2B A3 41 70 CD 1A 42 85 00 43 07 00
AE 57 A3 41 90 0F 1D 42 96 00 43 07 00

16 F6 B5 41 20 1B 2C 42 01 00 4F 07 00
0D 0C B6 41 20 1B 2C 42 01 00 4F 07 00
0D 0C B6 41 C0 AD 2B 42 02 00 4F 07 00

5B 7E AB 41 30 91 0A 42 00 00 4A 07 00
6D 52 AB 41 00 99 0A 42 01 00 4A 07 00
76 3C AB 41 30 91 0A 42 01 00 4A 07 00
76 3C AB 41 F0 71 0A 42 01 00 4A 07 00
5B 7E AB 41 F0 71 0A 42 01 00 4A 07 00
76 3C AB 41 00 99 0A 42 00 00 4A 07 00
76 3C AB 41 C0 79 0A 42 00 00 4A 07 00

Over Fire
44 43 A9 41 C0 49 12 42 F9 00 45 07 00
20 9B A9 41 C0 49 12 42 D1 00 45 07 00
20 9B A9 41 10 07 14 42 D1 00 45 07 00
05 DD A9 41 00 57 15 42 0A 00 45 07 00
05 DD A9 41 30 C6 16 42 03 00 45 07 00
EA 1E AA 41 60 35 18 42 02 00 45 07 00
D8 4A AA 41 60 35 18 42 04 00 45 07 00
D8 4A AA 41 20 FE 1B 42 02 00 45 07 00
9A E4 AA 41 20 FE 1B 42 03 00 45 07 00
49 AA AB 41 60 82 20 42 01 00 45 07 00
49 AA AB 41 80 B2 25 42 02 00 45 07 00
7B A3 AD 41 00 D3 2A 42 03 00 45 07 00
FD D6 AE 41 F0 0A 30 42 04 00 45 07 00
FD D6 AE 41 D0 9E 34 42 04 00 45 07 00
26 E6 B0 41 60 EC 38 42 02 00 45 07 00
26 E6 B0 41 50 AD 3C 42 02 00 45 07 00
97 45 B2 41 50 AD 3C 42 03 00 45 07 00
97 45 B2 41 E0 00 40 42 03 00 45 07 00
1A 79 B3 41 50 8F 41 42 02 00 45 07 00
C9 3E B4 41 30 AC 44 42 1F 01 45 07 00
BC D1 B6 41 30 9A 47 42 B6 00 45 07 00
BC D1 B6 41 B0 4F 49 42 86 00 45 07 00
48 EF B7 41 B0 4F 49 42 21 00 45 07 00
48 EF B7 41 80 51 4A 42 AC 00 45 07 00

Trying to catch the shutoff
AC B8 AA 41 A0 72 F8 41 00 00 46 07 00
A3 CE AA 41 A0 72 F8 41 00 00 46 07 00
9A E4 AA 41 A0 72 F8 41 00 00 46 07 00
9A E4 AA 41 E0 91 F8 41 00 00 46 07 00
AC B8 AA 41 E0 91 F8 41 00 00 46 07 00
AC B8 AA 41 C0 43 F8 41 00 00 46 07 00
B4 A2 AA 41 A0 72 F8 41 00 00 46 07 00
B4 A2 AA 41 90 A1 F8 41 00 00 46 07 00
Abrupt Change
CF 60 AA 41 80 A1 F8 41 00 00 82 00 00
CF 60 AA 41 E0 91 F8 41 00 00 82 00 00
B4 A2 AA 41 80 A1 F8 41 00 00 82 00 00
B4 A2 AA 41 40 82 F8 41 00 00 82 00 00
BD 8C AA 41 E0 91 F8 41 00 00 82 00 00
B4 A2 AA 41 00 63 F8 41 00 00 82 00 00
CF 60 AA 41 60 D0 F8 41 00 00 82 00 00
BD 8C AA 41 60 D0 F8 41 00 00 82 00 00
BD 8C AA 41 80 A1 F8 41 00 00 82 00 00
C6 76 AA 41 80 A1 F8 41 00 00 82 00 00
C6 76 AA 41 00 E0 F8 41 00 00 82 00 00
^Last Transmission^`

DieKatzchen, 2019-11-13:
Okay, so reversing the order and splitting it into 2 bytes, bytes, 4 bytes, 4 bytes gives me some useful data. That first 2 is 00 07 for "I am active" and 00 00 for "I am about to go to sleep". Next 4 bytes I can't figure out. they seem to barely change while I'm recording, but change (relatively) dramatically between readings.  Next 4 bytes stay relatively consistent, except they increased dramatically when held over a flame, so probably heat. Last 4 bytes are probably moisture by process of elimination. It increased slightly when held over heat, possibly because it was a gas flame (and therefore produced water vapor as a byproduct). I'll test more tonight by leaving it on the kitchen sink while I shower.

DieKatzchen, 2019-11-18:
So I've figured out one of the mystery bytes. The first (or last, I can't tell what order it's displaying in when I tell it to display in hex) is accelerometer data, but not any specific axis but rather summing them somehow. It might actually be two bytes, the byte next to it never seems to change, I might not be able to swing it hard enough to roll over to the next byte. Will have to test it on the spin cycle. I have some numbers from after the shower but I haven't had a chance to look at them yet.

1technophile, 2019-11-24:
Hello @DieKatzchen ,

Did you had the chance to finish your test and reverse engineering?

DieKatzchen, 2019-11-25:
I've been busy, but I ran it through the dryer just now. Turns out it keeps advertising even when the hub is up, so I was able to get some data points from both the official app and my BLE sniffer. I put it up on Imgur, I'll see if I can correlate the numbers tomorrow.

DieKatzchen, 2019-11-28:
Seems like no matter what I do, that first data point is a massive outlier. Since I only took 3 data points, that means I'm left with 2 which is as good as having none. I'll take some more data points later.

RussH, 2020-09-14:
@DieKatzchen did you ever progress this?

DieKatzchen, 2020-09-25:
I'm gonna be completely honest... I forgot about this. I did take a whole LOAD of data points, but never processed them. I'm gonna see what I can get out of it.

DigiH, 2022-08-04:
This has been implemented with the 0.9.15 release of OpenMQTTGateway

https://github.com/1technophile/OpenMQTTGateway/releases/tag/v0.9.15