1technophile/OpenMQTTGateway #492: make call

utya1988, 2019-11-13:
Hello. Can i make call using mqtt using 2g gateway? thanks
If not could you explan, how can i make it.

1technophile, 2019-11-14:
Hello,

Making a call is not currently supported (only sending and receiving SMS)

utya1988, 2019-11-14:
Maybe. I can change some lines ot the code to implement this feature?

1technophile, 2019-11-15:
Yes , here is the methods available on the library used
https://github.com/skorokithakis/A6lib/blob/master/A6lib.h

1technophile, 2019-11-15:
For my curiosity could you share the use case behind your request?

utya, 2021-07-05:
I have the gate for parking. For open the gate i should make the call.

1technophile, 2021-07-05:
You could use the SMS function of the library to trigger your gate opening or closing, also you may take a look to the library below:
https://github.com/vshymanskyy/TinyGSM

closed 2023-02-04 by 1technophile