1technophile/OpenMQTTGateway #493: Error compiling in VS

PaulAntonDeen, 2019-11-17:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
Assembler messages:
Fatal error: can't create .pio/build/nodemcuv2-all/lib432/FastLED_ID126/wiring.cpp.o: No such file or directory
*** [.pio/build/nodemcuv2-all/lib432/FastLED_ID126/wiring.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/nodemcuv2-all/lib432/FastLED_ID126/noise.cpp.o: No such file or directory
Assembler messages:
Fatal error: can't create .pio/build/nodemcuv2-all/lib432/FastLED_ID126/power_mgt.cpp.o: No such file or directory
*** [.pio/build/nodemcuv2-all/lib432/FastLED_ID126/noise.cpp.o] Error 1
*** [.pio/build/nodemcuv2-all/lib432/FastLED_ID126/power_mgt.cpp.o] Error 1

**To Reproduce**
Steps to reproduce the behavior:
Compiling on Mac with latest VS, OMG0.92 code with only this one uncommented:
default_envs = nodemcuv2-all
A similar problem also occurs when uncommenting:
;default_envs = nodemcuv2-ir 

No other modifications to the files.

**Expected behavior**
Compile successfully

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment (please complete the following information):**
OpenMQTTGateway V0.92

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2019-11-17:
Hello,

Thanks for the detailled bug report.
I have downloaded v0.9.2 and compiled successfully:
`default_envs = nodemcuv2-all`
&
`default_envs = nodemcuv2-ir`

The build status of v0.9.2 on Travis was also passing.

I'm suspecting an issue with your environment or with MacOS. Could you try to build another arduino program with PIO and VSC?
Alternatively you could try also to remove the .pio folder from OpenMQTTGateway folder.

PaulAntonDeen, 2019-11-17:
Removing .pio folder made no difference.
Do you have a recommendation for another project to try build?
I will try and get my hands on a windows machine and solve it that way.

1technophile, 2019-11-17:
I would begin by a simple project like a blink example.
But If you have a windows machine I would try first with it.

PaulAntonDeen, 2019-11-21:
Tried it on Windows 10, downlaoded a fresh copy of VSC, used the library to install platform.io and downloaded the 0.92 source.zip from Github. Opened up platformio.ini and uncommented default_envs = nodemcuv2-all.
Failures:
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\Paul\Downloads\OpenMQTTGateway-0.9.2\main\ZsensorBH1750.ino file
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\Paul\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\Paul\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
PermissionError: [WinError 5] Access is denied: 'c:\\Users\\Paul\\Downloads\\OpenMQTTGateway-0.9.2\\.pio\\libdeps\\nodemcuv2-all\\_tmp_installing-kt0kgnfx-package\\.git\\objects\\pack\\pack-b9f151d1eef4c2fad26c3829eae6f62f2e985fe8.idx':
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Paul\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Paul\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Paul\.platformio\platforms\espressif8266\builder\main.py", line 316:
    target_elf = env.BuildProgram()
  File "C:\Users\Paul\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Paul\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "C:\Users\Paul\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 578:
    rmtree(src)
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 516:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 400:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\Paul\.platformio\python37\lib\shutil.py", line 398:
    os.unlink(fullname)

1technophile, 2019-11-21:
Hmm, I had also this issue with the latest platform.io version, it has been corrected in v0.9.3beta.
If you take this version you should be safe.
Alternatively and if you don't need BH1750, just delete this file:
`C:\Users\Paul\Downloads\OpenMQTTGateway-0.9.2\main\ZsensorBH1750.ino`

PaulAntonDeen, 2019-11-22:
0.93beta works on Mac!

closed 2019-11-22 by PaulAntonDeen

1technophile, 2019-11-22:
great!