1technophile/OpenMQTTGateway #495: Led status pin NodeMCU

byte4geek, 2019-11-23:
Hi,

i have no problem whit the project, i have all working and correctly integrated with home assistant but...
...what pin can i use on NodeMCU to connect the leds to see the status?

i found the lines in the User_config.h

/*-------------DEFINE PINs FOR STATUS LEDs----------------*/
#ifdef ESP8266
  #define led_receive 40
  #define led_send 42
  #define led_info 44

but i dont know the related pins.

1technophile, 2019-11-24:
Hello,

You can take the pins that you don't use from this sheet:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1864843263

byte4geek, 2020-01-15:
Thanks, using you docs i configured the leds

closed 2020-01-15 by byte4geek