1technophile/OpenMQTTGateway #496: Statical IP instead of DHPC

byte4geek, 2019-11-25:
Hi,

i set the ip config in these line in the User_config.h but my mqttgateway always aobtain an ip by DHCP, maybe i missing someting?

```
// these values are only used if no dhcp configuration is available
const byte ip[] = { 192, 168, 1, 3 }; //ip adress
const byte gateway[] = { 192, 168, 1, 1 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte Dns[] = { 192, 168, 1, 11 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte subnet[] = { 255, 255, 255, 0 }; //ip adress
```

Thanks a lot

1technophile, 2019-11-26:
Hi,

Indeed you have to uncomment [this line](https://github.com/1technophile/OpenMQTTGateway/blob/2db4da014d63b3d3fad4abf11cedefcc0448130e/main/main.ino#L654) 
and comment the line above

I need to improve that...

byte4geek, 2019-11-29:
very thanks, now it works, is very usefull that this configuration is placed in User_config.h or a commnet that redirect to the line of interested file.

closed 2019-12-07 by 1technophile

1technophile, 2019-12-08:
I have integrated the change,
now you just have to uncomment:
`//#define NetworkAdvancedSetup true`

Would it be possible to test if this is working for you?