1technophile/OpenMQTTGateway #497: 请求添加ssd1306 Request to add ssd1306

671792320, 2019-11-25:
Thank you very much for developing this project. How can I request to add ssd1306 OLED support. Thank you

1technophile, 2019-11-26:
Hello,
What are the informations that you would like to display on the screen?

671792320, 2019-11-27:
> Hello,
> What are the informations that you would like to display on the screen?

Thank you for your reply

I want to display the received RF code and temperature and humidity code on the display,

If you can customize the display and call all the plug-in support under your project, it will be more perfect

1technophile, 2019-11-27:
> If you can customize the display and call all the plug-in support under your project, it will be more perfect

By plugin you mean the different activated gateways and sensors?

671792320, 2019-12-01:
> > If you can customize the display and call all the plug-in support under your project, it will be more perfect
> 
> By plugin you mean the different activated gateways and sensors?

Yes, I want to show some data on the OLED

AskAlice, 2021-10-08:
STILL WOULD LOVE THIS

1technophile, 2023-02-04:
It is now possible

closed 2023-02-04 by 1technophile