1technophile/OpenMQTTGateway #501: IOT standalone network

happytm, 2019-12-02:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
I wanted to create low power sensor network for coin cell powered devices.Using adhoc network protocol it is possible to have very low power consumption devices running on coin cell. 

**Describe alternatives you've considered**
I have mainly considered Lora (bit expensive) & ESPNow (require extra device to receive data). 

**Additional context**
See if it is possible to add an extra protocol option to your gateway.

I have been working on a simple very low power IOT application using uMQTT broker. It is a template for a standalone star network of few battery powered sensors using adhoc network protocol which can be customized easily.  There are 2 sketches ProbeSender and ProbeReceiver. ProbeReceiver is using uMQTT broker to connect to outside world and it is up all the time using mains power. All ProbeSendor devices are not directly connected to internet. I have implemented basic 2 way communication between ProbeReceiver and all ProbeSender device on the network. For issuing commands to ProbeSender devices on network I use android app called MQTT Dash. Only requirement for this code to work efficiently (less power consumption) is to have same access point channel as home's main access point across all devices. 

Is there any way you can implement it in your code ?

The link to code is below:

https://github.com/happytm/BatteryNode

Thanks.

1technophile, 2019-12-04:
Hello,

> Only requirement for this code to work efficiently (less power consumption) is to have same access point channel as home's main access point across all devices.

Could you explain more this constrain please?

happytm, 2019-12-04:
It means you have to  have fixed channel set on your home's main access point and then use same channel value for "apChannel" variable in both ProbeReceiver and ProbeSender sketches.This way it take less time and only scan for that channel.

But it is possible to change this behavior by replacing:  
        int n = WiFi.scanNetworks(true, false, apChannel, (uint8*) gateway);
with
        int n = WiFi.scanNetworks(true, false);

in ProbeSender sketch. So now it will scan all channels and thereby use longer time to send data.

Thanks.

1technophile, 2019-12-04:
ok, thanks for the infos.

Is the probre receiver usefull for people having already an MQTT broker?

happytm, 2019-12-04:
Yes. In that case there will be 2 brokers receiving  whatever they are subscribed to. Ideal situation would be to have define in code to have option to turn on/off either regular external broker or built in uMQTT broker.

Thanks.

1technophile, 2019-12-04:
I don't get the added value in this case of the uMQTT, you could replace the center of your star by your main MQTT broker?

happytm, 2019-12-04:
The only added value is that having a broker on the center of star whole network become truly standalone network without requiring MQTT broker on either local machine or in cloud. And yet it is possible to reach each device on  network by simply using app like MQTT Dash to issue commands or read & display sensor data from them.  My main goals for this project were low power devices  in truly standalone network. Power consumption wise it perform lot better than ESPNow.

Thanks

1technophile, 2019-12-04:
How much do you achieve in terms of power consumption ?

happytm, 2019-12-04:
A lot. If you send data with regular wifi & MQTT it will take 5 to 8 seconds at 80mAh. With ESPNow it will take about 200 milliseconds at 80mAh. With this code my test show it takes 60 milliseconds to send data. In theory it will run fine for one year one measurement of up to 5 sensors every 15min, from a 450mAh LiFepo4 according to this calculator at http://www.of-things.de/battery-life-calculator.php.

I am using it with coin cell LIR2450 with little solar panel https://www.ebay.com/itm/10pcs-5V-30mA-Micro-Mini-Power-Solar-Cells-Panel-Board-Set-For-DIY-Toy-53-30mm/352673192324?_trkparms=aid%3D555018%26algo%3DPL.SIM%26ao%3D1%26asc%3D61112%26meid%3D200eec1edfcf4fbcbb5b66d5e3fa5118%26pid%3D100005%26rk%3D3%26rkt%3D12%26mehot%3Dpf%26sd%3D333302663283%26itm%3D352673192324%26pmt%3D1%26noa%3D0%26pg%3D2047675&_trksid=p2047675.c100005.m1851 to hopefully run it without ever charging the battery from a charger.

Thanks.

closed 2020-02-09 by stale[bot]