1technophile/OpenMQTTGateway #508: ESP32 stop working after a few hours

frnandu, 2019-12-08:
I'm using OMG to transmit temperatures from a few Mijia sensors, and I'm getting this behavior (stops after a few hours, sometimes a few days).

I checked Troubleshooting section which says to comment the multiCore line on User_config.h, but the current version of the file doesn't have this line anymore.
How can I disable multicore on current development version to see if this helps with the problem?
Is there anything else I can try to fix it?

Thanks.

1technophile, 2019-12-08:
hello,

could you post the version used and the configuration (gateways and sensors)?

frnandu, 2019-12-08:
I'm using the development branch with only the ZgatewayBT.

1technophile, 2019-12-08:
Did you changed some settings on user_config.h or the interval between scans?

frnandu, 2019-12-08:
On user_config.h just changed to manual wifi with AP+password, and the MQTT config.
On config_BT.h I changed to:

#define TimeBtw_Read 45555 //define default time between 2 scans
#define Scan_duration 25 //define the time for a scan

Do you think these changes can have influence on the crashes?
I will try with v0.9.3beta and default delays to check if it works ok.

1technophile, 2019-12-08:
> Do you think these changes can have influence on the crashes?
Yes, unfortunately ESP32 is very sensitive regarding its BLE functions

> I will try with v0.9.3beta and default delays to check if it works ok.
Perfect

I have 2 users with several days of duration discussing [here](https://community.openmqttgateway.com/t/esp32-ble-gateway-dying-every-day-and-a-half/697/26?u=1technophile), maybe you can join this thread to report your results

frnandu, 2019-12-11:
Using 0.9.3beta with default delays working for 3 days now without a problem.

If changing BLE delays is so problematic on ESP32, maybe it should be some kind of warning in the documentation and/or (more importantly) on the config_BT.h to stronly disencourage changing them.
Anyway thanks for your wonderfull effort which is OMG ;)

1technophile, 2019-12-11:
I think you can change the TimeBtw_Read, it seems that Scan_duration is more sensitive.

In all the case I will add some comments in the code and in the docs

closed 2019-12-14 by 1technophile

frnandu, 2019-12-16:
After 5 days it freezes again :(

when I do:
mosquitto_sub -t home/# -v
i get:
home/OpenMQTTGateway/LWT offline
home/OpenMQTTGateway/version 0.9.3beta

I tried:
mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/set -m '{"interval":0}'
and then:
mosquitto_pub -t "home/OpenMQTTGateway/commands/MQTTtoSYS/set" -m '{"cmd":"restart"}'

but nothing.
What else can/should I check?
Trying to leave it connected to a computer on serial for a few days, hoping it will log something usefull to diagnose is a bit complicated.

I'm using the mijia temp sensors to controll heating in the house, so when this freezes in the middle of the night it means we start freezing literally!!! :)

It must be something I can change to make it work, otherwise my family will get quite angry with me :)