1technophile/OpenMQTTGateway #509: Ethernet.linkStatus() == LinkON and W5100 shield

bvduijn, 2019-12-08:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
In the master https://github.com/1technophile/OpenMQTTGateway/blob/54c141d925594e8ad0d75a3483f174328f9e1fd1/main/main.ino#L867 branch Ethernet linkstatus is wrong for W5100 shield. It seems to have been resolved in development.  https://github.com/1technophile/OpenMQTTGateway/blob/c0ddf77c9f5bce6f2d7500870e7dbef490c7f51b/main/main.ino#L947 Please merge this change.

**To Reproduce**
Steps to reproduce the behavior:
1. Install from master branch on Arduino Uno or Mega

**Expected behavior**
Please merge this change. It took me a few hours before i noticed this "simple" error

**Environment (please complete the following information):**
* OpenMQTTGateway version used V0.92, master

**Additional context**
Issue has already been addressed in dev branch but wrong config still persist in master

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2019-12-14:
I'm going to publish soon the v0.9.3, it will include this change

1technophile, 2020-01-17:
v0.9.3 published

closed 2020-01-17 by 1technophile