1technophile/OpenMQTTGateway #511: Change of publish function

bvduijn, 2019-12-12:
**Describe the bug**
Hi @1technophile , I see the fuction to publish was changed in Zmqttdiscovery.ino https://github.com/1technophile/OpenMQTTGateway/blob/c0ddf77c9f5bce6f2d7500870e7dbef490c7f51b/main/ZmqttDiscovery.ino#L96 

In version 0.91 is was:
https://github.com/1technophile/OpenMQTTGateway/blob/c0e3c46a5bef665e301986dc767fbfa92a089a28/ZmqttDiscovery.ino#L398

Can you explain the reason for this change? Because of the new dir structure I cannot see the history properly.

Since version 0.92 I get errors like 

Unable to parse JSON DEEDBAFE5495gatewayRF: 'DEEDBAFE5495gatewayRF'
6:17 PM components/mqtt/discovery.py (WARNING) - message first occurred at 6:53 AM and shows up 341 times
Unable to parse JSON DEEDBAFE5495gatewayRF: 'gatewayRF'
6:17 PM components/mqtt/discovery.py (WARNING) - message first occurred at 6:53 AM and shows up 341 times
Unable to parse JSON DEEDBAFE5495gatewayRF: 'home/OpenMQTTGateway/433toMQTT'
6:17 PM components/mqtt/discovery.py (WARNING) - message first occurred at 6:53 AM and shows up 341 times
Unable to parse JSON DEEDBAFE5495humdht: '%'
6:17 PM components/mqtt/discovery.py (WARNING) - message first occurred at 6:53 AM and shows up 341 times
Unable to parse JSON DEEDBAFE5495humdht: '{{ value_json.hum }}'
6:17 PM components/mqtt/discovery.py (WARNING) - message first occurred at 6:53 AM and shows up 341 times

In HA discovery.

**To Reproduce**
Steps to reproduce the behavior:
1. Install version 0.92 from Master branch
2. Use MQTT discovery in Home Assistant

**Expected behavior**
Just a valid JSON message for discovery 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used V0.92, master.
* ZMqttDiscovery.ino

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2019-12-14:
`pub_custom_topic`

Enable to publish to a particular topic, compared to 

`pub`

which publish to the matin topic set into user_config.h or wifi manager.

This change was introduced in v0.9.3beta so as to enable to set gateway name and base topic into wifi manager.

1technophile, 2020-01-02:
I close, the issue, if you have other question don't hesitate to post it on the [forum](https://community.openmqttgateway.com).

1technophile, 2020-01-02:
Closed to fast maybe, I will recheck the discovery of DHT

1technophile, 2020-01-11:
rechecked and OK

closed 2020-01-11 by 1technophile