1technophile/OpenMQTTGateway #513: MQTT Topic Mijia Thermostat

jrockstad, 2019-12-22:
I reccive 3 different messages from my Mijia thermostats with the same MQTT topic:

```
home/OpenMQTTGateway_ESP8266_BLE/BTtoMQTT/[MAC]
{
  "hum" : 52.9,
  "tem" : 22.6
}
```

```
home/OpenMQTTGateway_ESP8266_BLE/BTtoMQTT/[MAC]
{
  "batt" : 52
}
```

```
home/OpenMQTTGateway_ESP8266_BLE/BTtoMQTT/[MAC]
{
  "id" : "AA:AA:AA:AA:AA:AA",
  "rssi" : -86,
  "distance" : 16.54559,
  "servicedata" : "***"
}
```

Would it be posible to change the topics to something like this?

```
home/OpenMQTTGateway_ESP8266_BLE/BTtoMQTT/[MAC]/sensor
{
  "hum" : 52.9,
  "tem" : 22.6
}
```

```
home/OpenMQTTGateway_ESP8266_BLE/BTtoMQTT/[MAC]/batt
{
  "batt" : 52
}
```

```
home/OpenMQTTGateway_ESP8266_BLE/BTtoMQTT/[MAC]/state
{
  "id" : "AA:AA:AA:AA:AA:AA",
  "rssi" : -86,
  "distance" : 16.54559,
  "servicedata" : "***"
}
```

1technophile, 2019-12-22:
Hello, could you explain the reason behind this need ?

jrockstad, 2019-12-22:
When Home Assistant read one MQTT message the attributes from the last message become's blank. I'm sure I can store the values in template sensors, turn of json to have different topics for all values or something else, but the way I see it the easiest would be to have different MQTT topics :)

1technophile, 2019-12-22:
I think you can do it an easier way; just add `is_defined` on your value template:
`"{{ value_json.tem | is_defined }}"`

jrockstad, 2019-12-23:
Well, that worked perfectly! Thanx! 😄 
I did update the wiki under Mijia thermometer to make it easier for others.
You do as you see fit with this request

1technophile, 2019-12-23:
Great, the wiki is moved progressively to http://docs.openmqttgateway.com
if you want you can also propose a PR for this document:
https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/integrate/home_assistant.md

jrockstad, 2019-12-23:
Done 😄

1technophile, 2019-12-23:
Thanks ! To finish the work you need to submit a pull request from your branch.

1technophile, 2019-12-28:
PR done

closed 2019-12-28 by 1technophile

nikito7, 2020-06-22:
Still is a problem with MQTT Dash Android

How use defined with jsonpath?
To get values I use
$.tem

How use 'if ($.tem)' alike ?

But I receive *tem* and *hum* messages separated

*tem* msg will clear *hum* and vice-versa

Maybe an android app problem

But change topic to home/***/MAC/tem will be a better fix, I think

I never seen tem/hum in same message

1technophile, 2020-06-22:
Hi,

Could you open a topic into:
https://community.openmqttgateway.com/
Regarding MQTT Dash Android to discuss with the community about this