1technophile/OpenMQTTGateway #516: Wifi disconnected

byte4geek, 2019-12-28:
Hi, im running 0.9.3rc wiht RF receiver and DHT

on platformio.ino i enabled:
default_envs = nodemcuv2-rf

sometimes i find the nodeMCU halted with this continuous message on serial:

wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected

when i press reset button the system restart correctly.
Is it possible to do an automatic restart in this case?

1technophile, 2019-12-28:
Hi,

If you wait for several minutes, it does not reconnect automaticaly?

byte4geek, 2019-12-30:
i'm wait for 30mins and no reconnect.

dandresm, 2019-12-30:
Hi, same here. I have an esp32 with 0.9.3rc version and DHT, RF, RF2 and BLE enabled on it.
[image]
In my case, this behaviour comes from previous version.

1technophile, 2020-01-02:
> i'm wait for 30mins and no reconnect.

I have uploaded to a nodemcu the nodemcu-v2-rf env (v0.9.3rc).
I have stopped my wifi and here is the result from the serial monitor:
```
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
wifi disconnected
MQTT connection...
Connected to broker
Subscription OK to the subjects
Uptime (s)
234
```

It does reconnect automatically when I restart the wifi. I reproduced this behavious several times.

I also added a DHT to the NodeMCU and I also get automatic wifi reconnection:
```
Pub json into:
home/OpenMQTTGateway_ESP8266_TEST/DHTtoMQTT/dht1
{"hum":51.8,"temp":19.1}
wifi disconnected
wifi disconnected
wifi disconnected
MQTT connection...
Connected to broker
Subscription OK to the subjects
```

Could you indicate me the board used and if you are uploading by PIO or other method?

byte4geek, 2020-01-02:
I'm using AZ-DELIVERY NodeMcu V3
im uploading the firmware using the usb port on board and the Flahs Download Tool 2.6.5.
the firmware is self compiled by ATOM.

1technophile, 2020-01-02:
> the firmware is self compiled by ATOM
By self compiled you mean that you compile it yourself with stock parameters of platformio.ini and user_config.h files?

byte4geek, 2020-01-10:
Yes

1technophile, 2020-01-11:
I have a nodemcu v3 I will try with it

1technophile, 2020-01-17:
@byte4geek do you still see wifi disconnection without automatic reconnection?

byte4geek, 2020-02-24:
I change my old Router with a Fritz!Box, more stable, this issue is not longer present

marrobHD, 2020-04-06:
After rebooting my sonoff rf bridge it won´t connect back to WiFi.
It shows at the bottom:
`Not Connected to JK12NoT_Oben`
[image]

1technophile, 2020-04-06:
Maybe wait 120s for the portal to come off, if you're access point hasn't change it should reconnect

closed 2020-06-12 by stale[bot]