1technophile/OpenMQTTGateway #517: Invert status led's

m4rcu5, 2019-12-30:
Hi,

Related to issue #221, which had the request for a status LED on the Sonoff RF Bridge. I would like to request the option to invert the LED.

The `WIFI_LED` on the `433RFBRDIGE R2` is connected to `GPIO13`. Defining `led_info 13` will indeed light the LED.
However it seems to be inverted. Lighting most of the time and turning off during transmission.

Would it be possible to implement a inverse option on the 3 leds, like the `#define LED1_PIN_INVERSE` in the Espurna code?

1technophile, 2019-12-30:
Hi,

Yes of course

1technophile, 2020-08-26:
Hi @m4rcu5 

I have just pushed the requested [change](https://github.com/1technophile/OpenMQTTGateway/pull/745), if you have some time to test it, it would be great

closed 2020-08-27 by 1technophile

m4rcu5, 2020-08-29:
Thanks @1technophile I can confirm the LED is behaving as the docs say, and the inverted mode works.

1technophile, 2020-08-29:
Perfect, thanks for the feedback!