1technophile/OpenMQTTGateway #518: SRFB with release 0.9.3rc results in watchdog reset

m4rcu5, 2019-12-30:
Uploading the 0.9.3rc release, prebuild binary from the releases page, to a SRFB R2 v1.0 will results in a `Soft WDT reset` when trying to publish the json discovery topics.

The following output can be obtained from the serial terminal:

```
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.xx.149
connected...yeey :)
saving config
{"mqtt_server":"192.168.xx.161","mqtt_port":"1883","mqtt_user":"","mqtt_pass":"","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_SRFB"}OpenMQTTGateway mac: 
84:0D:8E:66:14:32
OpenMQTTGateway ip: 
192.168.xx.149
1883
Connecting to MQTT by IP adress
192.168.xx.161
ZgatewaySRFB setup done 
Serial Baud19200
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects
Uptime (s)
120
omgStatusDiscovery
Pub json discovery into:
homeassistant/binary_sensor/840D8E661432/config
{"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"OpenMQTTGateway_SRFB","uniq_id":"840D8E661432","dev_cla":"connectivity","pl_on":"Online","pl_off":"Offline","pl_avail":"Online","pl_not_avail":"Offline"}
Pub json discovery into:
homeassistant/switch/840D8E661432restart/config
{"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"restart OMG","uniq_id":"840D8E661432restart","pl_on":"{\"cmd\":\"restart\"}","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"home/OpenMQTTGateway_SRFB/commands/MQTTtoSYS/config","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.3rc","identifiers":["840D8E661432"]}}
Pub json discovery into:
homeassistant/switch/840D8E661432erase/config
{"stat_t":"home/OpenMQTTGateway_SRFB/LWT","name":"erase OMG","uniq_id":"840D8E661432erase","pl_on":"{\"cmd\":\"erase\"}","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"home/OpenMQTTGateway_SRFB/commands/MQTTtoSYS/config","device":{"name":"OpenMQTTGateway_SRFB","manufacturer":"OMG_community","sw_version":"v0.9.3rc","identifiers":["840D8E661432"]}}

Soft WDT reset
```

Followed by a stack trace.

If I compile a version myself from #development, currently at 86f3286ed129d7657a2c7e04884ad6d82a21acba, the issue does not occur.

1technophile, 2019-12-30:
Hello, When you compile it yourself could you point out the env or config used ?

m4rcu5, 2019-12-30:
Of course:

```
OpenMQTTGateway % git --no-pager diff 
diff --git a/main/User_config.h b/main/User_config.h
index 9ad957f..2c27fb0 100644
--- a/main/User_config.h
+++ b/main/User_config.h
@@ -175,7 +175,7 @@ char gateway_name[parameters_size * 2] = Gateway_Name;
 #endif
 #ifndef led_info
   #ifdef ESP8266
-    #define led_info 44
+    #define led_info 13
   #elif ESP32
     #define led_info 44
   #elif __AVR_ATmega2560__ //arduino mega
@@ -236,4 +236,4 @@ char gateway_name[parameters_size * 2] = Gateway_Name;
 /*-------------------ACTIVATE TRACES----------------------*/
 #define TRACE 1  // commented =  trace off, uncommented = trace on
 
-#endif
\ No newline at end of file
+#endif
diff --git a/platformio.ini b/platformio.ini
index 9be3be7..fe18006 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -29,7 +29,7 @@ extra_configs =
 
 ;default_envs = sonoff-basic-rfr3
 ;default_envs = sonoff-basic
-;default_envs = rfbridge
+default_envs = rfbridge
 ;default_envs = esp32dev-all
 ;default_envs = esp32dev-rf
 ;default_envs = esp32dev-pilight
@@ -78,7 +78,7 @@ wifimanager = tzapu/WiFiManager#c3ff582
 ethernet = Ethernet
 esp8266_mdns = esp8266_mdns
 wire = Wire
-fastled = FastLED
+#fastled = FastLED
 
 [env]
 framework = arduino

OpenMQTTGateway % platformio run --environment rfbridge
```

```
OpenMQTTGateway % platformio --version
PlatformIO, version 4.1.0

OpenMQTTGateway % platformio lib list
Library Storage: ./OpenMQTTGateway/.pio/libdeps/rfbridge
ArduinoJson
===========
#ID: 64
An elegant and efficient JSON library for embedded systems

Version: 5.13.4
Homepage: https://arduinojson.org/?utm_source=meta&utm_medium=library.json
Keywords: json, rest, http, web
Compatible frameworks: arduino
Compatible platforms: atmelavr, atmelsam, espressif32, espressif8266, intel_arc32, microchippic32, nordicnrf51, nordicnrf52, ststm32, teensy, timsp430
Authors: Benoit Blanchon

PubSubClient
============
#ID: 89
A client library for MQTT messaging. MQTT is a lightweight messaging protocol ideal for small devices. This library allows you to send and receive MQTT messages. It supports the latest MQTT 3.1.1 protocol and can be configured to use the older MQTT 3.1 if needed. It supports all Arduino Ethernet Client compatible hardware, including the Intel Galileo/Edison, ESP8266 and TI CC3000.

Version: 2.7
Keywords: ethernet, mqtt, m2m, iot
Compatible frameworks: arduino
Compatible platforms: atmelavr, espressif8266
Authors: Nick O'Leary

WifiManager
===========
WiFi Configuration manager with web configuration portal for ESP boards

Version: c3ff582
Keywords: wifi, wi-fi, esp, esp8266, esp32, espressif8266, espressif32, nodemcu, wemos, arduino
Compatible frameworks: arduino
Compatible platforms: espressif8266, espressif32
Authors: tzapu, tablatronix
Source: git+https://github.com/tzapu/WiFiManager#c3ff582

esp8266_mdns
============
#ID: 1091
mDNS queries and responses on esp8266. Or to describe it another way: An mDNS Client or Bonjour Client library for the esp8266.

Version: b7c88fda89
Keywords: esp8266, mdns, zeroconf, bonjour, avahi
Compatible frameworks: arduino
Compatible platforms: espressif8266
Authors: dunk
```

I had to comment the `fastled` library, as it will no longer compile with the latest `platform-espressif8266`. This would reference https://github.com/FastLED/FastLED/issues/733
Are you already aware of this?

1technophile, 2019-12-31:
> I had to comment the fastled library, as it will no longer compile with the latest platform-espressif8266. This would reference FastLED/FastLED#733
> Are you already aware of this?

No, thanks for the info.

Thanks for the details. Maybe a track to follow are the difference between test_config.h and user_config.h.

m4rcu5, 2020-01-01:
I checked the differences between the `Test_config.h` and `User_config.h`. 

My main suspect is this difference:
```diff
--- main/User_config.h	2019-12-30 19:39:19.219390124 +0100
+++ test/Test_config.h	2019-12-30 19:30:15.943910939 +0100
@@ -215,25 +160,22 @@
 #define eraseCmd            "erase"
 
 // define if we concatenate the values into the topic
-//#define valueAsASubject true
+#define valueAsASubject true
 
 //variables to avoid duplicates
 #define time_avoid_duplicate 3000 // if you want to avoid duplicate mqtt message received set this to > 0, the value is the time in milliseconds during which we don't publish duplicates
 
+#ifdef ESP32
+  //#define multiCore //uncomment to use multicore function of ESP32 for BLE
+#endif
 #if defined(ESP8266) || defined(ESP32) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)
-  #define JSON_MSG_BUFFER 512 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
+  #define JSON_MSG_BUFFER 1024 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
   #define ARDUINOJSON_USE_LONG_LONG 1
 #else // boards with smaller memory
   #define JSON_MSG_BUFFER 64 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
 #endif
```

I have changed the `JSON_MSG_BUFFER 1024` in my `User_config.h` as well, and now my own compiled version also hangs and throws a stacktrace when the `SYStoMQTT` topic is supposed to be published.

It seems that a value `1024` is indeed to high for an `ESP8255`

1technophile, 2020-01-01:
>I have changed the JSON_MSG_BUFFER 1024 in my User_config.h as well, and now my own compiled version also hangs and throws a stacktrace when the SYStoMQTT topic is supposed to be published.
It seems that a value 1024 is indeed to high for an ESP8255

Thanks a lot for this analysis!
:-)
We need to update the test_config.h to align it with user_config.h or simply remove it as is it not anymore needed with PIO Continuous Integration.

closed 2020-01-02 by 1technophile