1technophile/OpenMQTTGateway #522: mqtt sever configuration not work

douzs, 2020-01-06:
hi,i find a bug,and it can reproduce. When i input the configuration in wifimanager,router ssid and password,mqtt username and password,and mqtt server ip and port.
It can connect the router .but in the log ,it still connect 192.168.1.17
the log:

*WM: [2] . 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.2.62
connected...yeey :)
saving config
failed to open config file for writing
{"mqtt_server":"192.168.1.17","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"your_password","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP8266_RF2"}OpenMQTTGateway mac: 
3C:71:BF:38:E2:7B
OpenMQTTGateway ip: 
192.168.2.62
1883
Connecting to MQTT by IP adress
192.168.1.17
RF_EMITTER_PIN 
3
RF_RECEIVER_PIN 
0
ZgatewayRF2 setup done 
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
failure_number

i use nodemcuv2-rf-firmware.bin

i hope you can help me fix the bug.If it is not a bug,plese tell me how to do.
i need the gateway and i think it is a great thing that you create it.
thank you

1technophile, 2020-01-06:
Yes, If you take v0.9.3rc you should not have this bug

nvschilleman, 2020-01-08:
I have a similar issue using 0.9.3rc. Wifi connection is succesful, but no connection to broker. My MQTT broker shows no connection attempts.

fragolinux, 2020-01-12:
same here (using a wemos esp32 and then a doit esp32 to test if it was the board itself), using "ALL" edition of firmware, latest 0.9.3... going from the ALL to the BLE it connected without issues...

douzs, 2020-01-13:
> 我在使用0.9.3rc时遇到了类似的问题。Wifi连接成功，但未连接到代理。我的MQTT代理未显示任何连接尝试。

It is important to point that if mqtt fails to connect after sone times attempts, the wifi will be reset

1technophile, 2020-01-13:
> going from the ALL to the BLE it connected without issues...

Thanks for the info, i will check if there is issues with the env containing all modules.
In the meantime please chose the env with only what you need.

henkiejan1, 2020-01-19:
Sorry for cross-post but i posted my reply on a closed issue. I got the same problem. Really everything tried but continuesly the same error. It is about ESP32_BLE.

I want to connect with password but it gives continuesly a rc=5 error. When i change the password i got a connection reset by peer 104 error. I have changes anonymous allow on and off. the name of the password file changes, password changes. a few times flashed again differen devices but nothing works... then i tries a demo sketch mqtt esp with pubsub client on the same device. And that works directly correct. Finally looks like that 0.92Beta correct works...

1technophile, 2020-01-19:
> I want to connect with password but it gives continuesly a rc=5 error. When i change the password i got a connection reset by peer 104 error. I have changes anonymous allow on and off.

Doesn't seem to be the same issue as the one described here :-(
Please go to http://community.openmqttgateway.com

closed 2020-03-26 by stale[bot]