1technophile/OpenMQTTGateway #526: ZgatewayBT:453:49: error: 'class BLEAdvertisedDevice' has no member named 'getServiceDataCount'

jjtoard, 2020-01-12:
Hello,

I tried almost everything I think to get the openmqttgateway installed on my esp32 but don't succeed in it.

Error I get at this point is :
ZgatewayBT:453:49: error: 'class BLEAdvertisedDevice' has no member named 'getServiceDataCount'

Can someone helpe me fixing this

1technophile, 2020-01-12:
Hi,

This is because you are not using the good BLE library. You should take the one from the release page  : 
https://github.com/1technophile/OpenMQTTGateway/releases/download/v0.9.3/esp32dev-ble-libraries.zip

jjtoard, 2020-01-12:
now it compiles till the end but I get an error that the scetch is to large
I also tried to remove all the .ino and all the .h file except for the user_config.h, config_BT.h, ZgatewayBT.ino but then I get errors again

1technophile, 2020-01-12:
Yes, indeed. You need to select into your arduino ide menu the min spiffs partition so as to free space for the program.

jjtoard, 2020-01-12:
what board should I use then

1technophile, 2020-01-12:
You need to choose esp32dev and select min spiffs in flash or partition scheme sub menu

jjtoard, 2020-01-12:
first compilation, lots of errors,
second compilation worked fine
could finally upload the code
testing will be for tomorrow
thanks a lot for helping.

Could the code on github be updated.  I can imagine I'm not the only one having this issues.

Again thanks a lot for the help.

1technophile, 2020-01-13:
Yes indeed, docs will be updated.

closed 2020-03-21 by stale[bot]