1technophile/OpenMQTTGateway #528: Trigger button, Pressed Button Held, Erasing ESP Config, restarting (in a loop)

byte4geek, 2020-01-14:
Hi,

im trying the release 0.9.3 but after i set the wifi info on the WEB interface, on serial i can see a restart with: 

**Trigger button Pressed
Button Held
Erasing ESP Config, restarting**

i reconnect on WEB interface reset all data and the issue occur again and again
this issue is not present on 0.9.3rc

i have a AZ-Devlivery NodeMCU v3 with RF module and DHT

Static ip

**SERIAL LOG:**

mounting FS...
mounted file system
Adv wifi cfg
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: name
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP: 192.168.1.3
*WM: [2] Custom static IP/GW/Subnet/DNS
*WM: [2] Custom STA IP/GW/Subnet
*WM: [1] STA IP set: 192.168.1.3
*WM: [3] WIFI station disconnect
*WM: [1] No saved credentials, skipping wifi
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED
*WM: [2] AccessPoint set password is VALID
*WM: [1] 8passer8
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP8266_RF
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC started
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] WiFi Scan ASYNC completed in 2195 ms
*WM: [2] WiFi Scan ASYNC found: 3
*WM: [3] -> connectivitycheck.gstatic.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> clients3.google.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> clients3.google.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> c.whatsapp.net
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> connectivitycheck.gstatic.com
*WM: [2] <- Request redirected to captive portal
*WM: [2] <- HTTP Root
*WM: [3] -> 192.168.4.1
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [2] Scan is cached 7313 ms ago
*WM: [3] -> 192.168.4.1
*WM: [2] <- HTTP Wifi
*WM: [2] Scan is cached 8378 ms ago
*WM: [1] 3 networks found
*WM: [2] DUP AP: HOME-NET
*WM: [2] AP:  -31 HOME-NET
*WM: [2] AP:  -80 GUEST-NET
*WM: [3] _staShowStaticFields
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [3] Sent config page
*WM: [3] -> connectivitycheck.gstatic.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> connectivitycheck.gstatic.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> connectivitycheck.gstatic.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> connectivitycheck.gstatic.com
*WM: [2] <- Request redirected to captive portal
*WM: [3] -> portal.fb.com
*WM: [2] <- Request redirected to captive portal
*WM: [2] <- HTTP Root
*WM: [3] -> 192.168.4.1
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [2] Scan is cached 15123 ms ago
*WM: [2] Portal Timeout In 109 seconds
*WM: [2] <- HTTP WiFi save
*WM: [3] Method: POST
*WM: [2] Parameters
*WM: [2] --------------------
*WM: [2] server: 192.168.1.2
*WM: [2] port: 1883
*WM: [2] user: mqttuser
*WM: [2] pass: 12345678
*WM: [2] name: OpenMQTTGateway_ESP8266_RF
*WM: [2] topic: home/
*WM: [2] --------------------
*WM: [3] static ip: 192.168.1.3
*WM: [3] static gateway: 192.168.1.1
*WM: [3] static netmask: 255.255.255.0
*WM: [3] Sent wifi save page
*WM: [2] process connect
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP: 192.168.1.3
*WM: [2] Custom static IP/GW/Subnet/DNS
*WM: [2] Custom STA IP/GW/Subnet
*WM: [1] STA IP set: 192.168.1.3
*WM: [3] WIFI station disconnect
*WM: [1] Connecting to new AP: HOME-NET
*WM: [3] Using Password: 12345678
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS]
*WM: [1] Got IP Address:
*WM: [1] 192.168.1.3
Should save config
*WM: [2] disconnect configportal
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting
connected...yeey :)
saving config
{"mqtt_server":"192.168.1.2","mqtt_port":"1883","mqtt_user":"mqttuser","mqtt_pass":"12345678","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP8266_RF"}OpenMQTTGateway mac:
4C:11:AE:0A:C6:2B
OpenMQTTGateway ip:
192.168.1.3
1883
Connecting to MQTT by IP adress
192.168.1.2
RF_EMITTER_PIN
4
RF_RECEIVER_PIN
5
RF setup ok
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
Trigger button Pressed
Button Held
Erasing ESP Config, restarting
Formatting requested, result:
▒
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
   mounting FS...
mounted file system
Adv wifi cfg
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: name
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP: 192.168.1.3
*WM: [2] Custom static IP/GW/Subnet/DNS
*WM: [2] Custom STA IP/GW/Subnet
*WM: [1] STA IP set: 192.168.1.3
*WM: [3] WIFI station disconnect
*WM: [1] No saved credentials, skipping wifi
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED
*WM: [2] AccessPoint set password is VALID
*WM: [1] 12345678
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP8266_RF
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC started
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] WiFi Scan ASYNC completed in 2193 ms
*WM: [2] WiFi Scan ASYNC found: 3

byte4geek, 2020-01-14:
to avoid wires connection issue i try to unplug all wires from board

byte4geek, 2020-01-14:
This is my User_config.h

`/*
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal  and a MQTT broker
   Send and receiving command by MQTT

  This program enables to:
 - receive MQTT data from a topic and send signals corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received signals

    Copyright: (c)Florian ROBERT

    This file is part of OpenMQTTGateway.

    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#ifndef user_config_h
#define user_config_h
/*-------------------VERSION----------------------*/
#define OMG_VERSION "version_tag"

/*-------------CONFIGURE WIFIMANAGER-------------(only ESP8266 & SONOFF RFBridge)*/
/*
 * The following parameters are set during the WifiManager setup process:
 * - wifi_ssid
 * - wifi_password
 * - mqtt_user
 * - mqtt_pass
 * - mqtt_server
 * - mqtt_port
 *
 * To completely disable WifiManager, define ESPWifiManualSetup.
 * If you do so, please don't forget to set these variables before compiling
 *
 * Otherwise you can provide these credentials on the web interface after connecting
 * to the access point with your password (SSID: WifiManager_ssid, password: WifiManager_password)
 */
/*-------------DEFINE GATEWAY NAME BELOW IT CAN ALSO BE DEFINED IN platformio.ini----------------*/
#ifndef Gateway_Name
  #define Gateway_Name "OpenMQTTGateway"
#endif

#define Base_Topic "home/"

/*-------------DEFINE YOUR  NETWORK PARAMETERS BELOW----------------*/

#define NetworkAdvancedSetup true //uncomment if you want to set advanced network parameters for arduino boards, not uncommented you can set the IP and mac only
#ifdef NetworkAdvancedSetup // for arduino boards advanced config
  #if defined(ESP8266)||defined(ESP32)
    const byte ip[] = { 192, 168, 1, 3 }; //ip adress of the gateway, already defined for arduino below
  #endif
  const byte gateway[] = { 192, 168, 1, 1 };
  const byte Dns[] = { 129, 168, 1, 2 };
  const byte subnet[] = { 255, 255, 255, 0 };
#endif

#if defined(ESP8266)||defined(ESP32)  // for nodemcu, weemos and esp8266
[... 154 lines cut ...]

// define if we concatenate the values into the topic
//#define valueAsASubject true

//variables to avoid duplicates
#define time_avoid_duplicate 3000 // if you want to avoid duplicate mqtt message received set this to > 0, the value is the time in milliseconds during which we don't publish duplicates

#if defined(ESP8266) || defined(ESP32) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)
  #define JSON_MSG_BUFFER 512 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
  #define ARDUINOJSON_USE_LONG_LONG 1
#else // boards with smaller memory
  #define JSON_MSG_BUFFER 64 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
#endif

#define TimeBetweenReadingSYS 120000 // time between system readings (like memory)
#define subjectSYStoMQTT  "/SYStoMQTT"
#define subjectMQTTtoSYSset "/commands/MQTTtoSYS/config"

//#define subjectTRACEtoMQTT "OpenMQTTGateway/log" //uncomment if you want to see traces on OpenMQTTGateway/log topic

/*-------------------ACTIVATE TRACES----------------------*/
#define TRACE 1  // commented =  trace off, uncommented = trace on

#endif
`

1technophile, 2020-01-14:
Could you try to change the trigger pin number?

```
#ifndef TRIGGER_PIN
#ifdef ESP8266
#define TRIGGER_PIN 14 // pin D5 as full reset button (long press >10s)
#elif ESP32
#define TRIGGER_PIN 0 // boot button as full reset button (long press >10s)
#endif
#endif
```

For the code formatting select your text and use the button "insert code". It will make the reading easier.

byte4geek, 2020-01-17:
Ok changing the trigger pin has solved the issue

1technophile, 2020-01-17:
as 13 is not the default pin I close the issue

closed 2020-01-17 by 1technophile

MunkeyBalls, 2020-02-29:
@1technophile I also ran into some problems after updating to 0.9.3
I use GPIOInput which for ESP8266 defaults to PIN14 (D5), but the same GPIO pin is bound to the reset trigger. Only after doing serial monitoring I noticed that instead of GPIOInput I got  only got the "Trigger button" message, and when triggering for a longer time the erase espconfig/restart. 

After changing the trigger pin to 13 everything works again. I'm wondering if this was an intentional change from your end, since it seems to break the GPIOInput working on the default config.

Thanks very much for your project

1technophile, 2020-03-09:
Nope, that's not intentional :-) 

I've just changed it to 13.

Thanks