1technophile/OpenMQTTGateway #529: Fail on compile with  #define ESPWifiManualSetup true, using ATOM

byte4geek, 2020-01-14:
I encounter an error when i try to compile release 0.9.3 using ATOM and WIFI info (siid, password) statical settings by uncomment the line 69 User_config.h  #define ESPWifiManualSetup true

c:/users/admin/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2-rf\src\main.ino.cpp.o:(.text._Z9MQTTtoS
YSPcRN11ArduinoJson10JsonObjectE+0x24): undefined reference to `setup_wifimanager(bool)'
c:/users/admin/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2-rf\src\main.ino.cpp.o: in function `MQT
TtoSYS(char*, ArduinoJson::JsonObject&)':
main.ino.cpp:(.text._Z9MQTTtoSYSPcRN11ArduinoJson10JsonObjectE+0xd2): undefined reference to `setup_wifimanager(bool)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2-rf\firmware.elf] Error 1
========================= [FAILED] Took 30.53 seconds =========================

 
Environment          Status    Duration
-------------------  --------  ------------
rfbridge             IGNORED
esp32dev-all         IGNORED
esp32dev-rf          IGNORED
esp32dev-pilight     IGNORED
esp32dev-ir          IGNORED
esp32dev-ble         IGNORED
esp32-m5stick-ble    IGNORED
esp32-m5stack-ble    IGNORED
esp32-m5stick-c-ble  IGNORED
ttgo-lora32-v1       IGNORED
nodemcuv2-all        IGNORED
nodemcuv2-ble        IGNORED
nodemcuv2-ir         IGNORED
nodemcuv2-rf         FAILED    00:00:30.531
rf-wifi-gateway      IGNORED
nodemcuv2-rf2        IGNORED
nodemcuv2-pilight    IGNORED
sonoff-basic         IGNORED
sonoff-basic-rfr3    IGNORED
atmega-all           IGNORED
uno-rf               IGNORED
uno-fastled          IGNORED
==================== 1 failed, 0 succeeded in 00:00:30.531 ====================

This my user config section:

/*-------------DEFINE YOUR  NETWORK PARAMETERS BELOW----------------*/

#define NetworkAdvancedSetup true //uncomment if you want to set advanced network parameters for arduino boards, not uncommented you can set the IP and mac only
#ifdef NetworkAdvancedSetup // for arduino boards advanced config
  #if defined(ESP8266)||defined(ESP32)
    const byte ip[] = { 192, 168, 1, 3 }; //ip adress of the gateway, already defined for arduino below
  #endif
  const byte gateway[] = { 192, 168, 1, 1 };
  const byte Dns[] = { 129, 168, 1, 2 };
  const byte subnet[] = { 255, 255, 255, 0 };
#endif

#if defined(ESP8266)||defined(ESP32)  // for nodemcu, weemos and esp8266
  #define ESPWifiManualSetup true //uncomment you don't want to use wifimanager for your credential settings on ESP
#else // for arduino boards
  const byte ip[] = { 192, 168, 1, 3 };
  const byte mac[] = {  0xDE, 0xED, 0xBA, 0xFE, 0x54, 0x95 }; //W5100 ethernet shield mac adress
#endif

#if defined(ESPWifiManualSetup) // for nodemcu, weemos and esp8266
  #define wifi_ssid "HOME-NET"
  #define wifi_password "12345678"
#endif

#define WifiManager_password "your_password" //this is going to be the WPA2-PSK password for the initial setup access point
#define WifiManager_ssid Gateway_Name //this is the network name of the initial setup access point
#define WifiManager_ConfigPortalTimeOut 120
#define WifiManager_TimeOut 5

grzegorz189, 2020-01-14:
807#endif
808
809#if defined(ESP8266) || defined(ESP32)

1technophile, 2020-01-14:
should be ok in development branch

byte4geek, 2020-01-15:
Thanks a lot now it works, this issue solved.

1technophile, 2020-01-15:
Great, thanks for the feedback!

closed 2020-01-15 by 1technophile