1technophile/OpenMQTTGateway #531: Feature request: add possibility to define MQTT server hostname/domain

untitledlt, 2020-01-16:
I use external MQTT server with external IP address. IP address sometimes changes so it would be nice to have possibility to specify MQTT server hostname instead of IP address.

1technophile, 2020-01-17:
you can do it [there](https://github.com/1technophile/OpenMQTTGateway/blob/aeb00c4a7a5b55fc04d41e2b0444d13c61085450/main/User_config.h#L94)

closed 2020-01-17 by 1technophile

untitledlt, 2020-01-17:
Oh cool! Should I leave `mqtt_server` blank on https://github.com/1technophile/OpenMQTTGateway/blob/aeb00c4a7a5b55fc04d41e2b0444d13c61085450/main/User_config.h#L104?

1technophile, 2020-01-17:
you can let it as it is, it will not be taken into account if you uncomment 
`//#define mqtt_server_name "www.mqtt_broker.com"`

1technophile, 2020-01-17:
You're welcome