1technophile/OpenMQTTGateway #535: xiomi miji parsing problems with latest update

Sansibari, 2020-01-19:
After updating the xiomi miji the values can't be parsed anymore (see attached printscreen form mqtt-explorer). Only the xiomi mi scale is parsed correct.

[image]

1technophile, 2020-01-19:
did you checked the history values of the mi jia to see if the sensor value were not reported previously?

Sansibari, 2020-01-19:
unfortunately I got no history data saved.

1technophile, 2020-01-19:
Could you indicate the firmware version of your mi jia?

Sansibari, 2020-01-20:
sent to you by email

1technophile, 2020-01-20:
Could you post it here maybe, I haven't received it?

Sansibari, 2020-01-23:
The firmware is 1.01.0066

Maybe for your interest:  I can see the device and the values in ioBroker via ble adapter:

[image]

If you need more information don't hesitate to request.

Thank you!

closed 2020-04-30 by stale[bot]