1technophile/OpenMQTTGateway #536: How to disable config messages

byte4geek, 2020-01-20:
Hi there is a way to disable the auto config message?
I prefer to configure manually the sensors on Home-Assistant

these message generate a mount of Traffic that i don't use.
Example of the message i want to disable:

` homeassistant/switch/4C11AE0AXXXXXX24704/config {"stat_t":"home/OpenMQTTGateway_ESP8266_RF/433toMQTT","name":"32024704","uniq_id":"4C11AE0ACXXXXXX4704","pl_on":"32000704","cmd_t":"home/OpenMQTTGateway_ESP8266_RF/commands/MQTTto433","device":{"name":"OpenMQTTGateway_ESP8266_RF","manufacturer":"OMG_community","sw_version":"version_tag","identifiers":["4C11AXXXXB"]}} `

Thanks a lot

1technophile, 2020-01-20:
Hello,

Yes , just remove the line below:
https://github.com/1technophile/OpenMQTTGateway/blob/4115f2a723f6d3d981cb7aaf42d52bd725417560/platformio.ini#L123

byte4geek, 2020-02-14:
thanks a lot it works fine

closed 2020-02-14 by byte4geek