1technophile/OpenMQTTGateway #537: Move version information from the version topic to SYStoMQTT

akasma74, 2020-01-21:
**Is your feature request related to a problem? Please describe.**
I think there is no point in publishing OMG version in a separated topic with a retain flag.
It just adds up to all that retained messages and makes little sense on its own.

**Describe the solution you'd like**
Just publish the version information alongside everything else like

> home/OMG_2nd_floor/SYStoMQTT {"version": "0.9.2", "uptime":3480,"freeMem":45352,"rssi":-56,"SSID":"wifi","ip":"192.168.0.37","mac":"61:C6:3A:EA:20:88","modules":"RF"}

1technophile, 2020-01-21:
good idea!

akasma74, 2020-03-22:
@1technophile any chance for it to be implemented?

1technophile, 2020-03-22:
yes, as it is easy it will be added to next version

closed 2020-03-23 by 1technophile