1technophile/OpenMQTTGateway #540: Support for Mi Jia LYWSD03

endrec, 2020-01-26:
Would it be possible to add [Mi Jia LYWSD03](https://www.aliexpress.com/item/4000481087115.html) support?

I did a quick search, and it seems that others are struggling to support it as well, as it does not seem to be obvious to extract measurements from the broadcasted data. :(

There is a working [Python script](https://github.com/JsBergbau/MiTemperature2), which connects to these devices and can read values.

jschaeke, 2020-02-06:
Indeed it isn't working with the current version. I am also interested as I've got 3 of them.

danielrheinbay, 2020-06-27:
Actually, they do seem to broadcast temperature and humidity in their ADV frames about ever 10 mins in bursts of >20 packets, albeit encrypted. Find details how to decrypt the ADV frames [here](https://github.com/Magalex2x14/LYWSD03MMC-info).
Implementation which relies on passive monitoring:
https://github.com/esphome/feature-requests/issues/552

orrpan, 2020-07-05:
> Actually, they do seem to broadcast temperature and humidity in their ADV frames about ever 10 mins in bursts of >20 packets, albeit encrypted. Find details how to decrypt the ADV frames [here](https://github.com/Magalex2x14/LYWSD03MMC-info).
> Implementation which relies on passive monitoring:
> [esphome/feature-requests#552](https://github.com/esphome/feature-requests/issues/552)

It is implemented in esphome dev branch now, tested it with three lywsd03mmc and it works well

Floyddotnet, 2020-07-05:
Currently, i didn't have any new update to the state of the LYWSD03 support.

It's hard to get working together with the BLE library. in my small test project, all works as expected. after integration this changes in the OMG source, i trigger many errors in the BLE library and after fixing most of then, the whole process crashes with really useless error messages or ble wifi coexistence errors.

1technophile, 2020-07-05:
Thanks for the update !  Despite the errors, did you succeed in retrieving the values in a readable format ?

michapr, 2020-07-11:
@Floyddotnet - Have you got to work it stable with multiple devices - without errors and reboots?
In my test project I can get all values - but after some time (loops) the BLE process is hanging (can not make connections more) and I get different error mesages with reboot in result...

Can you send me your working test project, if possible?

Thanks,
Michael

Floyddotnet, 2020-07-12:
Currently i'm on a business trip and not at home.

You can check my early work at: https://github.com/Floyddotnet/OpenMQTTGateway/commit/c2c0ca703986892ae7301d8273eca6e977c82983 (main/ZgatewayBT.ino)
and the branch: https://github.com/Floyddotnet/OpenMQTTGateway/commits/issue-296

I try to find my working test project remotely. if I found it, I will upload it asap.

Floyddotnet, 2020-07-12:
a small note, the above changes in main/ZgatewayBT.ino reports the correct temperature if the connection could be established.

michapr, 2020-07-13:
Hi, thanks.
I have tried it, without all the BLE changes - the only change I made is in FreeRTOS.cpp
```
//		rc = ::xSemaphoreTake(m_semaphore, portMAX_DELAY) == pdTRUE;
		rc = ::xSemaphoreTake(m_semaphore, 60000) == pdTRUE;

```
because the project was hanging if sensor did not respond more (far away).
But I still get in my small project the 
"Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled."
most after reading the characteristic for the LYWSD03MMC sensor.
 
Same I get in your project (without  the BLE changes).

Would be intersting to get a stable working project with the sensors only - will try later with your BLE library modifications.

Thanks!
Michael

michapr, 2020-07-13:
@Floyddotnet 
Additional - I get same errors as before with my simple project (without WiFi) and modified BLE lib with 3 LYWSD03MMC sensors.
Reading the values for a while, then hanging or reboot...
These are the same errors as in Gateway with modified BLE lib...

michapr, 2020-07-15:
@1technophile , @Floyddotnet 
I have tried to add the simple implementation from
https://github.com/karolkalinski/esp32-snippets
but cannot get to work it...
Additional, it will work only with low power mode, because a reboot will be needed for stable processing of the LYWSD03MMC sensor over a long time (we can find a lot of such comments in web, seems to be an issue in BLE lib).

If somebody have an idea how it can work - tell me please ;)
In this time I use a second ESP32 for gateway for these sensors.

Thanks!

1technophile, 2020-07-15:
Hi @michapr ,

Thanks for your investigations, we may try also to stop and restart the BLE component, but last time I tried that I didn't had so much chance with these functions.

Could you give me or point me a minimal code that retrieve the data and print it to the serial monitor?

Floyddotnet, 2020-07-15:
Hi @michapr, hi  @1technophile,

i'm sure that https://github.com/karolkalinski/esp32-snippets/blob/master/Mijia-LYWSD03MMC-Client/Mijia-LYWSD03MMC-Client.ino works on my esp32 an my LYWSD03MMC.

but you must adopte a fix for the ble libary: https://github.com/nkolban/esp32-snippets/issues/765#issuecomment-453900444

and you should doublecheck the length of pData. This must be exact 5, otherwise it can crash.

```
static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  float temp;
  float humi;
  Serial.print("Notify callback for characteristic ");
  Serial.println(pBLERemoteCharacteristic->getUUID().toString().c_str());

  if(length == 5) { // <-- this check is needed
    temp = (pData[0] | (pData[1] << 8)) * 0.01; //little endian 
    humi = pData[2];
    Serial.printf("temp = %.1f : humidity = %.1f \n", temp, humi);
  }

  pClient->disconnect();
}
```

Floyddotnet, 2020-07-15:
you can choise of one of the following fixes for BLEREmoteService.cpp:170

fix the type declaration to array
```
esp_gattc_char_elem_t result[10];
 while (true) { 
 	uint16_t count = 10; 
```
or decrease the count of chars be progresses to 1
```
esp_gattc_char_elem_t result;
 while (true) { 
 	uint16_t count = 1; 
```

i have checking my source https://github.com/Floyddotnet/OpenMQTTGateway/commit/4fa3aa6ccb86ab412e695db00947edfbf1cff475 and i see, i did the latter.

michapr, 2020-07-15:
Hi @Floyddotnet ,
the
```esp_gattc_char_elem_t result;
 while (true) { 
 	uint16_t count = 1; 
```
I had already inside. Added the length check (good idea ;) ...) But anyway the sketch is hanging after some loops here:
```
    void onDisconnect(BLEClient* pclient) {
      Serial.println("Disconnected"); 
```
(removing the restart...)

Even only with one sensor.
Would be interesting where is the difference between our "basic" projects.

Michael

1technophile, 2020-08-12:
Added [here](https://github.com/1technophile/OpenMQTTGateway/tree/BLEConnect) for testing purposes at this time.

michapr, 2020-08-13:
Hi,
thank you for this! I have tested it today.
At first remark - there is needed a new lib - NimBLE, should be installed before.
This lib **contain** the FreeRTOS.h/FreeRTOS.cpp, so there will be a conflict if FreeRTOS was installed before.

Then - all is working as expected (good job!), but the BLE problems above are the same.

Within 3,5 hours the ESP was restarted 15times:
```
...
09:08:29.580 -> N: Found 6 devices, scan end deinit controller
09:08:29.626 -> N: BLE Connect begin
09:08:33.630 -> N: Subject: /BTtoMQTT/A4C138C18C71
09:08:33.630 -> N: Received json : {"id":"A4:C1:38:C1:8C:71","model":"LYWSD03MMC","tempc":22.92,"tempf":73.256,"hum":58,"volt":2.292,"batt":19.2}
09:09:01.467 -> N: Subject: /BTtoMQTT/A4C1386BF6B2
09:09:01.467 -> N: Received json : {"id":"A4:C1:38:6B:F6:B2","model":"LYWSD03MMC","tempc":22.59,"tempf":72.662,"hum":58,"volt":2.46,"batt":36}
09:09:14.794 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
09:09:14.794 -> Core 0 register dump:
09:09:14.794 -> PC      : 0x4018fbc3  PS      : 0x00060a30  A0      : 0x800d3204  A1      : 0x3ffddc20  
09:09:14.794 -> A2      : 0x00000000  A3      : 0x51433f8d  A4      : 0x3ffc3ff8  A5      : 0x51433f8d  
09:09:14.794 -> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800ec63c  A9      : 0x3ffddbd0  
09:09:14.841 -> A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000007  A13     : 0x00000002  
09:09:14.841 -> A14     : 0x00000021  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
09:09:14.841 -> EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
09:09:14.841 -> 
09:09:14.841 -> Backtrace: 0x4018fbc3:0x3ffddc20 0x400d3201:0x3ffddc40 0x400d3724:0x3ffddcc0 0x4008dfad:0x3ffddce0
09:09:14.841 -> 
09:09:14.841 -> Rebooting...
...
```

It is not very "critical", because all will work again.
For 139 BLE connect procedures - 15 reboots.

Would be great to have a possibility to define the interval for these BLE connections, it will save battery when requesting every 10 or 15 minutes.

Anyway - a good step for usage of devices with BLE connection requirement !!  
THANKS!

1technophile, 2020-08-13:
Thanks for the feedback, I get also this error after a few hours, I'm going to check how we can avoid it.
And I will add also a separate timer for the BLEconnect process.

1technophile, 2020-08-13:
I have pushed a [set](https://github.com/1technophile/OpenMQTTGateway/commit/d62861d619e168a52158e5897f6598fa30be773c ) of modifications to the branch.
Running now without reset for 7h:
[image]

michapr, 2020-08-14:
Very good job!

in lines 506 and 595 seems to be a unnecessary ")":
`Log.trace(F("Callback process canceled by processLock" CR)));`

I have installed - and it is working stable until now.

1technophile, 2020-08-14:
> in lines 506 and 595 seems to be a unnecessary ")":
Log.trace(F("Callback process canceled by processLock" CR)));

Corrected thanks

I have [added](https://github.com/1technophile/OpenMQTTGateway/commit/1738b64cd1b5b7e0c5bb6db2865eabc4abe2339d) the number of scan before a connect variable `ScanBeforeConnect` in config_BT.h

michapr, 2020-08-14:
I have seen (while checking this feature) that it maybe not optimal, becase not every device will be found at every scan.
I have multiple LYWSD03MMC - and sometimes one will be found, sometimes two and sometimes all 3 together with other devices..
So if connect after 5 scans (as sample) - it will take the result of last scan, I think so? (not all devices)
Would it be possible to collect the devices while "ScanBeforeConnect" - or you already made it? ;)

1technophile, 2020-08-14:
> So if connect after 5 scans (as sample) - it will take the result of last scan, I think so? (not all devices)

No, it will take all the devices identified during the scans from the gateway start.

michapr, 2020-08-15:
I have tried, it is working fine! 

You can add the new sensor MHO-C401
It is a nice sensor (paper ink) for low price.
https://de.aliexpress.com/i/4001174769598.html

As I have seen it is (BLE-) compatible with LYWSD03MMC.
I have added it in code (as LYWSD03MMC) and it is working stable:

related logs:

```
N: Device detected: A4:C1:38:E6:41:F3
N: Received json : {"id":"A4:C1:38:E6:41:F3","rssi":-77,"distance":7.119795,"servicedata":"30588703fdf341e638c1a4280100"}

N: Received json : {"id":"A4:C1:38:E6:41:F3","model":"LYWSD03MMC","tempc":23.36,"tempf":74.048,"hum":68,"volt":2.814,"batt":71.4}
...

```
Detection:
(here must be inserted new model in code of course)
```
        Log.trace(F("Is it a MHO-C401?" CR));
      if (strstr(service_data, "588703") != NULL ) {
          Log.notice(F("MHO-C401 add to list for future connect" CR));
          if (device->sensorModel == -1)
            createOrUpdateDevice(mac, device_flags_init, LYWSD03MMC);
      }

```
servicedata lenght is all the time the same.

1technophile, 2020-08-18:
MHO-C401 added in the linked PR, thanks for the idea!

closed 2020-08-18 by 1technophile