1technophile/OpenMQTTGateway #541: Static ip ignored, DHCP ip obtained

byte4geek, 2020-01-31:
Hi with the paramiters configured as below the static ip was ignored and OMG receive ip from DHCP.
using ESP8266 on nodeMCU
version 0.94 rc

where i wrong?

`#define NetworkAdvancedSetup true //uncomment if you want to set advanced network parameters for arduino boards, not uncommented you can set the IP and mac only
#ifdef NetworkAdvancedSetup // for arduino boards advanced config
  #if defined(ESP8266)||defined(ESP32)
    const byte ip[] = { 192, 168, 1, 3 }; //ip adress of the gateway, already defined for arduino below
  #endif
  const byte gateway[] = { 192, 168, 1, 1 };
  const byte Dns[] = { 192, 168, 1, 2 };
  const byte subnet[] = { 255, 255, 255, 0 };
#endif

#if defined(ESP8266)||defined(ESP32)  // for nodemcu, weemos and esp8266
  #define ESPWifiManualSetup true //uncomment you don't want to use wifimanager for your credential settings on ESP
#else // for arduino boards
  const byte ip[] = { 192, 168, 1, 3 };
  const byte mac[] = {  0xDE, 0xED, 0xBA, 0xFE, 0x54, 0x95 }; //W5100 ethernet shield mac adress
#endif

#if defined(ESPWifiManualSetup) // for nodemcu, weemos and esp8266
  #define wifi_ssid "HOME-NET"
  #define wifi_password "12345678"
#endif

#define WifiManager_password "your_password" //this is going to be the WPA2-PSK password for the initial setup access point
#define WifiManager_ssid Gateway_Name //this is the network name of the initial setup access point
#define WifiManager_ConfigPortalTimeOut 120
#define WifiManager_TimeOut 5`

byte4geek, 2020-02-14:
Hi,
any update on this issue?

1technophile, 2020-02-20:
Hi, there was some issue regarding network setup with the release candidate, could you try with the v0.9.3?

byte4geek, 2020-02-24:
v0.9.3 do not have this issue, i wait for the next release, and for now i use OMG with dhcp IP

closed 2020-05-02 by stale[bot]