1technophile/OpenMQTTGateway #542: MiFlora - full JSON Data possible

Blackmore23, 2020-02-06:
Hi,

is it possible to get full JSON Data from the Flower Sensors frim Xiaomi?

like this:

BTtoMQTT
  C47C8xxxxx = {"lux":673, "tem":23.4, "fer": 555, "moi":234}

i know, the Data is read once per Connection, first lux, next try temp, etc...

but - the OpenMQTTGateway connect every time to the Broker - so it could read the Data, change the Data which is read, and send it back to the Broker...

it would be very easier to manage the full Data from the Broker...

Thanks a lot...

Björn

1technophile, 2020-02-07:
Hello,

For the moment the approach is to keep the transmitted values as minimal as possible.
This could lead to some warnings in the automation controller due to the fact that one time the temp will be published without the other elements and another time the hum without the others...
The gateway publish what it gets. The method used to retrieve ble data enables to get one value at a time (we listen to the broadcasted values by the BLE device). If we would like to publish all the values in one step we would need to store value in memory taking into account that they may have changed in the meantime ie :  publishing wrong values.
This kind of feature should be linked for me to another way of retrieving data :  by a BLE connection. I have done a first step toward that for miflora battery retrieving but I got a lot of instability generated by this function. I will retry later on.

Blackmore23, 2020-04-07:
Hi,

my Setup is:

Sensors, ESP32, MQTT-Broker in RPi, Wemos D1 Mini Wirth RGB Matrix Clock 64x196px

Thx Björn

closed 2020-06-13 by stale[bot]