1technophile/OpenMQTTGateway #543: Version tag shows as 'version_tag' when using development beach.

davebuk, 2020-02-06:
As per the title. I have built a binary using the development branch and prod_env.ini. The version tag doesn't display number, just the text 'version_tag'.

1technophile, 2020-02-07:
`version_tag` is currently set when producing the binaries and the code ziped into the release page, it is replaced by the tag value of the release. If you download the code from the source code tab of github indeed the version tag is not set but if you download it from the release page you should have it.
You can replace `version_tag` into `user_config.h` also.
At the moment I didn't found a way to have this automaticaly set with the developement branch.

davebuk, 2020-02-10:
That's fine. I guess there is an option within GitHub to add tags to the branch?

closed 2020-04-18 by stale[bot]