1technophile/OpenMQTTGateway #549: Extending supported protocols (protocol: 22)

TomHartogs, 2020-02-09:
Hi there,

I am trying to connect a remote control that has a protocol that does not seem to be supported yet. I understand that your focus might not be to integrate every protocol but can you perhaps point me in the right direction of where the protocols are defined and how I might extend them? Are the limitations in OpenMQTTGateway or in the RC-Switch library? In case it might help you, I have attached the serial monitor output of the signal.

`{"value":3428591,"protocol":22,"length":24,"delay":697}
Store val
Min ind: 
2
store code :
3428591/3568185
Col: val/timestamp
0:3428575/3545217
1:3428575/3545334
2:3428591/3568185
3:3428605/2425954
4:3428605/2425991
5:3428575/2428495
6:3428575/2428528
7:3428591/2436212
8:3428591/2454999
9:3428591/2455034
10:3428591/2474828
11:3428591/2496320
`

1technophile, 2020-02-10:
Hi,

If you see this:
`{"value":3428591,"protocol":22,"length":24,"delay":697}`
it seems that your protocol is supported. Could you indicate me what makes you feel it is not supported?

TomHartogs, 2020-02-11:
@1technophile Yes, it seems that the library is able to decode the signal. However, I was led to believe that I could not send messages using this protocol since it is not mentioned under supported protocols in [the documentation](https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF#send-data-by-mqtt-with-advanced-rf-parameters). 
Since the (RF) light fixture outside was not responding, I assumed this would be the reason. I will give it another try tonight to see if something else was perhaps wrongly configured.

1technophile, 2020-02-12:
Thanks for pointing that, we need to update this page.

If you are able to receive you should be able to send, if it doesn't work the first time maybe by tweaking the pulselength (delay)

1technophile, 2020-04-27:
I have added the following mention in the wiki page for info:
[_Doc for OMG >= v0.9.3 is now here_](http://docs.openmqttgateway.com)

closed 2020-04-27 by 1technophile