1technophile/OpenMQTTGateway #550: OMG not reconnecting to mqtt after broker restart

hallstig, 2020-02-11:
**Describe the bug**
I am running the builtin mqtt broker in Home assistant core and when restarting Home assistant the OpenMQTTGateway does not reconnect to the mqtt broker without me having to reset or power cycle it first.

**To Reproduce**
Steps to reproduce the behavior:
1. Use builtin mqtt broker in Home assistant core
2. Make broker unavailable for a while by restarting Home assistant
3. OpenMQTTGateway stays offline according to LWT and lack of other data getting published

**Expected behavior**
Reconnection to the mqtt broker after a given time, as this seems to have been the implemented fix for #378 and #398 describing the same issue.

**Screenshots**
Not really applicable.

**Environment (please complete the following information):**
* OpenMQTTGateway version 0.9.3 release (prebuilt nodemcuv2-rf-firmware.bin)

**Additional context**
Hardware is a genuine Wemos D1 mini pro v2.0.0 with a superheterodyne receiver (only receiver hooked up currently). Nothing else connected but power.

1technophile, 2020-02-12:
Hello,

> Make broker unavailable for a while

Could you give an idea if we are talking about minutes, tens or hours ?

hallstig, 2020-02-13:
The broker was down maybe a minute or so on these occasions, perhaps less. Just the normal business of restarting Home assistant when adding things.

I have hooked it up to serial now and stopped the broker, left it down for around 10 minutes but started seeing these on the console as soon as it went down.
```MQTT connection...
22:06:45.778 -> failure_number
22:06:45.778 -> 1
22:06:45.778 -> failed, rc=
22:06:45.778 -> -2
```
Started Home assistant again and it connected to MQTT instantly!
```
MQTT connection...
Connected to broker
22:15:40.835 -> Subscription OK to the subjects
```
For some reason Home assistant still shows the status sensor for OMG as offline. (I see it fine in MQTT Explorer) The motion sensor reports well tho.

I tried again but this time just restarted Home assistant as when I originally saw these issues, same behavior. OMG reconnects to MQTT just fine and the motion sensor reports just fine. However home assistant shows OMG as offline.

However if I restart the D1, it shows as online as soon as it has connected to MQTT again.

hallstig, 2020-02-17:
I noticed the other night that the D1 suddenly dropped off wifi just after posting some sensor data to MQTT. I have not really noticed this before so maybe just a fluke that it happened while I had the serial console up.
```
23:03:59.946 -> JSON str
wifi disconnected
23:04:19.979 -> wifi disconnected
wifi disconnected
23:04:40.011 -> wifi disconnected
wifi disconnected
23:05:00.045 -> wifi disconnected
```
It stayed off the wifi until after a power cycle.

ptempier, 2020-02-19:
I was going to open a ticket for this too.
In my case i poweroff HA/mosquito at night and powerup the morning (small single room appartement, didn't bought a server yet, can't decide/budget it) 
if i restart omg, it works fine afterward.

It is possible to set home assistant action at startup , so maybe have it send something that trigger a discover response (tasmota documented it in their wiki)?
or maybe if mosquito is offline for a long time, make retry with increased length, like every minutes for the first 30 minutes, every 5 minutes for the first 6hours, and every 15 minutes after that ?

1technophile, 2020-02-20:
> It is possible to set home assistant action at startup , so maybe have it send something that trigger a discover response (tasmota documented it in their wiki)?

You can restart the gateway by an mqtt command :
https://docs.openmqttgateway.com/use/gateway.html

> I was going to open a ticket for this too.
In my case i poweroff HA/mosquito at night and powerup the morning (small single room appartement, didn't bought a server yet, can't decide/budget it)
if i restart omg, it works fine afterward.

I need to do some test with the embeded HA broker

1technophile, 2020-04-16:
Should be resolved with the last development branch..
Don't hesitate to add feedback on this issue.

closed 2020-04-16 by 1technophile