1technophile/OpenMQTTGateway #551: MQTTtoONOFF bug

a-roz, 2020-02-15:
Board: SONOFF-BASIC-RFR3 hacked as described

Been used precompiled version with RF module from GitHub , 
and also compiled version with recommended
```
  '-DZactuatorONOFF="ONOFF"'
  '-DZsensorGPIOInput="GPIOInput"'
  '-DACTUATOR_ONOFF_PIN=12'
  '-DGPIOInput_PIN=0'
  '-DRF_RECEIVER_PIN=4'
  '-DZgatewayPilight="Pilight"'
```

Pilight and GPIOInput (pin 0) they are work as expected

When I've try publish
`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoONOFF -m OFF
`

I get response on serial
```
Hey I got a callback
MQTTtoONOFF
pin number
0
Pub ack into:
home/RFR3/ONOFFtoMQTT
```
but it should be 12, not 0. it means - actuator doesn't work
To check this I've put following to **ZactuatorONOFF.ino** line 67
`    int pin = 12;
`
and it works as expected.
Please correct this.

Also, need a MQTT message or parameter which contains Actuator state like
{ power: ON }

1technophile, 2020-02-20:
could you indicate the version you are using?

a-roz, 2020-02-20:
the latest 0.9.3 downloaded week ago

1technophile, 2020-02-27:
Thanks for the feedback

1technophile, 2020-06-19:
corrected

closed 2020-06-19 by 1technophile