1technophile/OpenMQTTGateway #554: Is MDNS enabled by default?

agoralive, 2020-02-18:
Not a bug/request but more a question.

Is mdns enabled by default within all the releases?

I can't seem to find this information anywhere :(

1technophile, 2020-02-20:
it is not enable by default , the parameter is [here](https://github.com/1technophile/OpenMQTTGateway/blob/8842ea1f4b7d277245307ee2b3699692ccf09658/main/User_config.h#L86)

closed 2020-02-20 by 1technophile

agoralive, 2020-02-25:
Then I need to compile myself in order to get at version with mdns enabled?

Or will OMG be able to connect to at MQTT broker using "server.local" even if this parameter is not enabled?

1technophile, 2020-02-29:
> Then I need to compile myself in order to get at version with mdns enabled?

Yes  you need to compile it yourself

agoralive, 2020-03-10:
Thank you for the answer