1technophile/OpenMQTTGateway #556: Zactuator  "undefined reference to `MQTTtoONOFF"

Djang0Pepper, 2020-02-21:
I've previoulsy tested other feature DHT,DS1820 working so easy !!
but with Zactuator :(
I've only decommented in User_config.h this line :
#define ZactuatorONOFF "ONOFF"    //ESP8266, Arduino, ESP32,  Sonoff RF Bridge

And I get 

_> ketch\main.ino.cpp.o: In function `MQTTtoSYS(char*, ArduinoJson::JsonObject&)':
> 
> C:\Users\fr3d\Documents\PlatformIO\Projects\OpenMQTTGateway_new\OpenMQTTGateway\main/main.ino:303: undefined reference to `MQTTtoONOFF(char*, char*)'
> 
> sketch\main.ino.cpp.o: In function `variantAsInteger<long unsigned int>':
> 
> C:\Users\fr3d\Documents\PlatformIO\Projects\OpenMQTTGateway_new\OpenMQTTGateway\main/main.ino:303: undefined reference to `MQTTtoONOFF(char*, char*)'
> 
> collect2.exe: error: ld returned 1 exit status_

Re-tested on a freshen redownloaded.

Djang0Pepper, 2020-02-21:
solution :
#ifdef ZactuatorONOFF // **outside the jsonpublishing** macro due to the fact that we need to use simplepublishing with HA discovery
      MQTTtoONOFF(topicOri, jsondata);
      #endif      
    #endif
If I put it inside i can compile.

1technophile, 2020-04-16:
Hi,

Strange, I've tried to uncomment `#define ZactuatorONOFF ` in User_config.h but I did not get any build error.
May you give me the board used?

thomas-merzenich, 2020-05-10:
I have the same error:
`/home/thomas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino_build_562085/sketch/main.ino.cpp.o:/home/thomas/Arduino/OpenMQTTGateway-development/main/main.ino:1323: undefined reference to MQTTtoONOFF(char*, char*)`

`/home/thomas/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino_build_562085/sketch/main.ino.cpp.o: in function receivingMQTT(char*, char*):`

`/home/thomas/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:166: undefined reference to MQTTtoONOFF(char*, char*)`

`collect2: error: ld returned 1 exit status`

I'm using LOLIN(WEMOS) D1 mini Pro.

david-j84, 2020-06-07:
I am also having the same issue with a WEMOS D1 Mini, I have experimented with different versions of the ArduinoJSON library as some limited troubleshooting. I tried the 0.9.3 and the 0.9.4b beta.

1technophile, 2020-06-07:
Could you share your `user_config.h` or/and your env definition please?

1technophile, 2020-06-10:
reproduced with Arduino IDE (works with PIO)

closed 2020-06-10 by 1technophile

halk1982, 2022-01-20:
Hello 1technophile, can't thank you enough for OMG project! It's absolutely amazing from several aspects. Regarding this issue, to my understanding this will be resolved in version 0.9.9, is that correct ?

1technophile, 2022-01-20:
Hello;

It should be resolved from v0.9.4, are you still getting this error?

halk1982, 2022-01-20:
It could be that I'm facing another issue then, apologies. I tried to upload to an ESP32 (Espressif) both BLE & Actuator ONOFF but failed with "Zactuator "undefined reference to `MQTTtoONOFF" error message. Then I tried to upload only ZActuatorONOFF to an ESP8266 and again failed with these messages:

[image]

P.S. I have already successfully created an ESP8266 with CC1101 and an ESP32 with BLE only.

1technophile, 2022-01-20:
Which version are you trying to build @halk1982 ?

halk1982, 2022-01-20:
The latest one, 0.9.8. 
After all, I managed to do it without any issue by using PlatformIO. 
Not sure what was the problem with Arduino IDE.