1technophile/OpenMQTTGateway #558: Unable to change Gateway_Name

maxi1134, 2020-02-26:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
I modify Gateway_Name variable to contain "Bedroom"
But the gateway_name remains at "OpenMQTTGateway_ESP32_BLE" once compiled.

**To Reproduce**
Open a fresh copy of the project.

Uncomment "default_envs = esp32dev-ble" in platformio.ini

Set line 51 of User_config.h to "  #define Gateway_Name "Bedroom" "

**Expected behavior**
The Gateway_name to be Bedroom.

**Screenshots**
https://imgur.com/a/oxFp4mJ

**Environment (please complete the following information):**
 OpenMQTTGateway version used:  Latest (0.93?)

1technophile, 2020-02-26:
If using with platformio ide you should change it on this file.
Search for esp32dev-ble and you should find all the parameters for this environment and especialy the gateway name.

maxi1134, 2020-02-26:
Thanks for the speedy answer!

When you say "If using with platformio ide you should change it on this file."

Which file are you referring to? I do not find a file called "esp32dev-ble".

maxi1134, 2020-02-26:
Update:

I did find this:

`platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DGateway_Name="OpenMQTTGateway_ESP32_BLE"'`

And replaced it with:

`platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DGateway_Name="Bedroom"'`

But still not working :(

maxi1134, 2020-02-26:
Update, tried with a second esp32, this one works on the first try... Starting to think that my ESP32 was in read-only or something.

1technophile, 2020-02-27:
> Update, tried with a second esp32, this one works on the first try... Starting to think that my ESP32 was in read-only or something.

It is possible that the first gateway name was stored on memory and recovered automaticaly.

1technophile, 2020-04-16:
I'm closing this issue, I you reproduce it don't hesitate to reopen

closed 2020-04-16 by 1technophile