1technophile/OpenMQTTGateway #559: ESP32-Cam - Can I use this board?

agoralive, 2020-02-28:
Headline kinda says it all, but....

Can I use the ESP32-Cam (OV2640) Board with OMG?

Direct link to AliExpress (example, no affiliate): https://www.aliexpress.com/item/33027303390.html?spm=a2g0o.productlist.0.0.5da940d6YnOIi2&algo_pvid=4a130f0f-1815-485c-8180-3ba958356cf5&algo_expid=4a130f0f-1815-485c-8180-3ba958356cf5-0&btsid=0be3746c15828686315401788e527f&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

I know that the camera does not work...... Just asking regarding the board.

jmw6773, 2020-05-01:
Yes, you can use it. I'm using it with the HCSR501 (PIR), HTU21 (temp/humidity), Bluetooth (presence detection), and ADC (for light levels) modules. (I also turn the built-in flashlight LED (GPIO4) on when the PIR detects motion.)

To use the ADC on this board, you will need to de-solder a tiny LED and then run a wire from that LED's pad to the what you're wanting to connect it to. For me, I also de-soldered the resistors that go to the VCC pin, making it an unused pinheader, and soldered the wire to it. This gave me easy access to A5 on the ESP32 chip.

1technophile, 2020-06-07:
I'm closing the issue, if you have other comment to add, feel free to reopen it.

closed 2020-06-07 by 1technophile