1technophile/OpenMQTTGateway #561: Wifi connection dropped and unable to reconnect during initial setup

danobot, 2020-03-01:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
Wifi connection is established and then immediately disconnected. See logs below.
```
*WM: [2] <- HTTP WiFi save
*WM: [3] Method: POST
*WM: [2] Parameters
*WM: [2] --------------------
*WM: [2] server: 192.168.1.17
*WM: [2] port: 1883
*WM: [2] user: your_username
*WM: [2] pass: your_password
*WM: [2] --------------------
*WM: [3] Sent wifi save page
*WM: [2] process connect
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] Connecting to new AP: unifi_not
*WM: [3] Using Password: strongpassword
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS]
*WM: [1] Got IP Address:
*WM: [1] 10.1.1.59
Should save config
*WM: [2] disconnect configportal
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting
connected...yeey :)
saving config
{"mqtt_server":"192.168.1.17","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"your_password"}OpenMQTTGateway mac:
84:F3:EB:B3:9F:76
OpenMQTTGateway ip:
10.1.1.59
1883
Connecting to MQTT by IP adress
192.168.1.17
Bosch BME280 Initialized - Result of .begin(): 0x0
No TSL2561 detected
TSL2561 Initialized. Printing detials now.
------------------------------------
Sensor:       TSL2561
Driver Ver:   1
Unique ID:    12345
Max Value:    17000.00 lux
Min Value:    1.00 lux
Resolution:   1.00 lux
------------------------------------

_2G_TX_PIN
12
_2G_RX_PIN
13
Init 2G module:
14

```
**To Reproduce**
Steps to reproduce the behavior:
1. Flash firmware to nodemcu.
2. Use wifi mangaer to add wifi credentials
3. Click save.

I can see in modem that the device connected and it was assigned an IP address.
I am confused what this `Bosch BME280 Initialized - Result of .begin(): 0x0` means.

After the Init 2G module line, the device is stuck waiting for a connection.
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used v0.9.3

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

danobot, 2020-04-30:
This is still an issue  I havent been able to resolve.

jmw6773, 2020-04-30:
For the BME280 sensor, 0x0 means that it couldn't find the sensor. It either isn't attached, is attached incorrectly, or is broken.

The initialization of the 2G modem blocks until a the modem returns that a 2G connection is made. I don't have a 2G module, but I'd recommend checking the pins you have connected and the _2G_MODULE_BAUDRATE variable to make sure it matches what you're modem uses. (Defaults to 9600, but the A7 might use a BAUD rate of 115200.)

Did you connect the 2G model to a FTDI (USB to Serial) adapter to see the output of the modem to verify the BAUD rate and that the modem is connecting before trying to use it with your OpenMQTTGateway?

danobot, 2020-04-30:
This is a NodeMCU connected to a FTDI adapter. The wifi module on the NodeMCU works fine.
THe problem happens when the FTDI adapter is connected or not. (only connected it to capture the logging output).

I do not have any sensors connected to the nodemcu, this is a clean flash of the firmware v0.9.3, no customisations.
After connecting to the WiFi manager and entering network credentials, this issue occurs where it connects, receives an IP address (as per log above) and then disconnects.
What is this Bosch BME280 and where is it defined?

jmw6773, 2020-04-30:
Okay, I think I understand now.

The firmware you flashed has at least the BME280, TSL2561, and the 2G functions enabled. The 2G function blocks until it gets a cellular connection, which is why you are seeing everything stop.

You should choose the firmware that matches the sensors/modules you are wanting to connect. If you are just wanting to flash to play around with it, try flashing the [nodemcuv2-ir-firmware.bin](https://github.com/1technophile/OpenMQTTGateway/releases/download/v0.9.3/nodemcuv2-ir-firmware.bin) 

It won't block if you don't have anything connected. And assuming you connect to an MQTT server, it will allow you to attach a simple an LED with a resistor to the D4 (GPIO2) pin of your nodemcu to see it flash when you publish to the "/commands/MQTTtoIR" topic. (If you don't connect to an MQTT service, the OpenMQTTGateway will erase it's preferences and restart into the WiFi manager.)

danobot, 2020-05-01:
Tried with `nodemcuv2-rf-firmware.bin` and `nodemcuv2-ir-firmware.bin`. Both were unsuccessful.
The log below is for the `ir` version:

```
mounted file system
reading config file
opened config file
{"mqtt_server":"tower.local","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP8266_IR"}
                                                                                                                                                            parsed json
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: name
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to saved AP: wifi_name
*WM: [3] Using Password: ****
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [2] 5000 ms connectTimeout set
*WM: [2] 5000 ms timeout, waiting for connect...
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 10.1.1.92
connected...yeey :)
saving config
{"mqtt_server":"tower.local","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP8266_IR"}OpenMQTTGateway mac: 
84:F3:EB:B3:AA:06
OpenMQTTGateway ip: 
10.1.1.92

 ets Jan  8 2013,rst cause:4, boot mode:(3,2)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
   mounting FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"tower.local","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP8266_IR"}
                                                                                                                                                            parsed json
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: name
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to saved AP: wifi_name
*WM: [3] Using Password: ****
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [2] 5000 ms connectTimeout set
*WM: [2] 5000 ms timeout, waiting for connect...
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] Connection result: WL_DISCONNECTED
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] your_password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP8266_IR
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2193 ms
*WM: [2] WiFi Scan ASYNC found: 13
*WM: [2] Portal Timeout In 95 seconds
*WM: [2] Portal Timeout In 65 seconds
```

jmw6773, 2020-05-01:
On the first pass, it connected to your wireless network, but then couldn't connect to the MQTT server. (I'm assuming, but it doesn't really say...) Is your MQTT server accessible from the 'unifi_iot' network? The IP would probably work more reliably than a '.local' dns name.

On the seconds pass, it couldn't connect to the 'unifi_iot' network and times out after 5sec of trying. How far are you from your AP?

danobot, 2020-05-02:
Ok, I will test again using the original wi-fi network with Mqtt configured and using the ir firmware.
The AP is right next to it.

1technophile, 2020-05-03:
Hi,

did you tried to erase the flash before uploading?
[Reboot loop ](https://docs.openmqttgateway.com/upload/troubleshoot.html#esp-continuous-restart-or-strange-behaviour)are caused most of the time by corrupted flash informations.

1technophile, 2020-05-03:
Hmm forget my last comment, I don't think it comes from this.

danobot, 2020-05-03:
Hi yes, I did erase the flash.
I will repeat the previous with a different NodeMCU device.

danobot, 2020-05-05:
It worked with another nodemcu.

closed 2020-05-05 by danobot